arduino/arduino-cli #446: Pololu A-Star fails compilation

inky699, 2019-10-17:
## Bug Report

### Current behavior
$ arduino-cli compile --fqbn arduino:samd:nano_33_iot a-star.ino 
Sketch uses 11812 bytes (4%) of program storage space. Maximum is 262144 bytes.
$ arduino-cli compile --fqbn pololu-a-star:avr:a-star32U4 a-star.ino
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x68 pc=0x7db81e]

goroutine 1 [running]:
github.com/arduino/arduino-cli/arduino/cores.(*PlatformRelease).GetLibrariesDir(0x0, 0xae0bc0)
	/__w/arduino-cli/arduino-cli/arduino/cores/cores.go:210 +0x3e
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.(*LibrariesManager).AddPlatformReleaseLibrariesDir(0xc001f55040, 0x0, 0x2)
	/__w/arduino-cli/arduino-cli/arduino/libraries/librariesmanager/librariesmanager.go:143 +0x43
github.com/arduino/arduino-cli/legacy/builder.(*LibrariesLoader).Run(0x12b5548, 0xc000032800, 0x12b5548, 0x0)
	/__w/arduino-cli/arduino-cli/legacy/builder/libraries_loader.go:62 +0x469
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x12b5548, 0xc000032800, 0x12b5548, 0x40c0f8)
	/__w/arduino-cli/arduino-cli/legacy/builder/container_setup.go:60 +0x314
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000032800, 0xc0001cf2d8, 0x20, 0x20, 0x414301, 0xe001cf1f0, 0x5555555555555555)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc0001cf530, 0xc000032800, 0xd37140, 0xc000427ba0)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:117 +0xd28
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:226
github.com/arduino/arduino-cli/commands/compile.Compile(0xd32080, 0xc0000c2000, 0xc0001cfcd0, 0xd21980, 0xc0000ac008, 0xd21980, 0xc0000ac010, 0xc0000ecf30, 0x9d4400, 0xb311a0, ...)
	/__w/arduino-cli/arduino-cli/commands/compile/compile.go:172 +0x1473
github.com/arduino/arduino-cli/cli/compile.run(0xc0001a9680, 0xc0001d04e0, 0x1, 0x3)
	/__w/arduino-cli/arduino-cli/cli/compile/compile.go:96 +0x280
github.com/spf13/cobra.(*Command).execute(0xc0001a9680, 0xc0001d0450, 0x3, 0x3, 0xc0001a9680, 0xc0001d0450)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2ae
github.com/spf13/cobra.(*Command).ExecuteC(0x128be80, 0xa8f291, 0xc0001cff88, 0xc00003e1d8)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fc
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:28 +0x2e

Simple code works for a Nano 33, but not for an install Pololu A-star
$ arduino-cli board listall 
Board Name                          FQBN                                      
4D Systems gen4 IoD Range           esp8266:esp8266:gen4iod                   
Adafruit Circuit Playground Express arduino:samd:adafruit_circuitplayground_m0
Arduino NANO 33 IoT                 arduino:samd:nano_33_iot                  
Arduino Primo                       arduino:nrf52:primo                       
Arduino Primo Core                  arduino:nrf52:primo_core                  
Arduino Tian                        arduino:samd:tian                         
Arduino Tian (MIPS Console port)    arduino:samd:tian_cons                    
Arduino Uno WiFi Rev2               arduino:megaavr:uno2018                   
Arduino Zero (Native USB Port)      arduino:samd:arduino_zero_native          
Arduino Zero (Programming Port)     arduino:samd:arduino_zero_edbg            
Digistump Oak                       esp8266:esp8266:oak                       
ESPDuino (ESP-13 Module)            esp8266:esp8266:espduino                  
ESPectro Core                       esp8266:esp8266:espectro                  
Pololu A-Star 328PB                 pololu-a-star:avr:a-star328PB             
Pololu A-Star 32U4                  pololu-a-star:avr:a-star32U4        

It was installed with:
arduino-cli.yaml:
board_manager:
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
    - https://files.pololu.com/arduino/package_pololu_index.json
$arduino-cli core update-index
$arduino-cli core search pololu
$arduino-cli core install pololu-a-star:avr

### Expected behavior
It should compile the code for the A-star board.

### Environment

$ arduino-cli version
arduino-cli Version: 0.5.0 Commit: 3be2287

(checkout out of git today (17th October 2019)
- OS and platform: $ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"

### Additional context
As seen above, it will compile on at least one other platform, including esp8266, which is installed the same way as the Pololu.

inky699, 2019-10-17:
Forgot to add the little program:
int analogValue = 0;    // variable to hold the analog value

void setup() {
	// open the serial port at 9600 bps:
	Serial.begin(9600);
}

void loop() {
	// read the analog input on pin 0:
	analogValue = analogRead(0);

	// print it out in many formats:
	Serial.println(analogValue);       // print as an ASCII-encoded decimal
	Serial.println(analogValue, DEC);  // print as an ASCII-encoded decimal
	Serial.println(analogValue, HEX);  // print as an ASCII-encoded hexadecimal
	Serial.println(analogValue, OCT);  // print as an ASCII-encoded octal
	Serial.println(analogValue, BIN);  // print as an ASCII-encoded binary
	delay(1000);
} // loop()

intrigus, 2019-10-26:
The pololu [package_pololu_index.json](https://files.pololu.com/arduino/package_pololu_index.json) does not define any tool dependencies.
I had a similar problem with another board.
You probably have to define which tools it needs for compilation.
Otherwise the cli does not know that it is missing libraries, assumes they exist and triggers an exception.

cmaglie, 2022-10-12:
I retried with the latest CLI:

```
$ arduino-cli core install pololu-a-star:avr
Sto scaricando i pacchetti...
pololu-a-star:avr@5.1.0 downloaded                                                                                                                                                                         
Sto installando la piattaforma pololu-a-star:avr@5.1.0...
Configurazione della piattaforma....
La piattaforma pololu-a-star:avr@5.1.0 è installata

$ arduino-cli compile -b pololu-a-star:avr:a-star32U4 test.ino
Lo sketch usa 4062 byte (14%) dello spazio disponibile per i programmi. Il massimo è 28672 byte.
Le variabili globali usano 153 byte (5%) di memoria dinamica, lasciando altri 2407 byte liberi per le variabili locali. Il massimo è 2560 byte.

Piattaforma utilizzata Versione Percorso                                                          
pololu-a-star:avr      5.1.0    /home/megabug/.arduino15/packages/pololu-a-star/hardware/avr/5.1.0
arduino:avr            1.8.5    /home/megabug/.arduino15/packages/arduino/hardware/avr/1.8.5      
```

A lot of improvements have been done to the Arduino CLI, and it seems to compile fine now, so I'm optimistically going aheah and closing the issue.
If the problem persists feel free to reopen the issue.

closed 2022-10-12 by cmaglie