arduino/arduino-cli #452: brew installation referenced in README fails on OS X Catalina

kevinclark, 2019-10-24:
The README directs install on OS X via 'brew install arduino-cli'. I've installed and removed twice and I'm seeing :

```
athena:fogger kevinclark$ brew update && brew install arduino-cli
Already up-to-date.
==> Downloading https://homebrew.bintray.com/bottles/arduino-cli-0.5.0.catalina.bottle.1.tar.gz
Already downloaded: /Users/kevinclark/Library/Caches/Homebrew/downloads/b7861ec6828871819319ae1c02d981b0d10bc2eadf0574d6383d18c8d84b5a6f--arduino-cli-0.5.0.catalina.bottle.1.tar.gz
==> Pouring arduino-cli-0.5.0.catalina.bottle.1.tar.gz
🍺  /usr/local/Cellar/arduino-cli/0.5.0: 5 files, 15.9MB
athena:fogger kevinclark$ arduino-cli 
-bash: /usr/local/bin/arduino-cli: cannot execute binary file
```

I checked out type, but got this odd output:

```
athena:fogger kevinclark$ type /usr/local/bin/arduino-cli
/usr/local/bin/arduino-cli is /usr/local/bin/arduino-cli
```

I'm on 10.15 (19A602).

```
athena:fogger kevinclark$ brew --version
Homebrew 2.1.15
Homebrew/homebrew-core (git revision e267a; last commit 2019-10-23)
Homebrew/homebrew-cask (git revision 74822d; last commit 2019-10-24)
```

Let me know if this is for this repo or needs to be taken up with a homebrew recipe. Figure if it's the official install path you'd want to know about it.

kevinclark, 2019-10-24:
Note that installing with the `--build-from-source` option does appear to work. It's just the bottle.

gvarisco, 2019-10-24:
Change requested in https://github.com/Homebrew/homebrew-core/pull/45756

masci, 2019-10-29:
0.6.0 is now available on homebrew, closing

closed 2019-10-29 by masci