arduino/arduino-cli #455: ESP32 compile seems to require working esptool

skandragon, 2019-10-25:
## Bug Report

### Current behavior
`arduino-cli compile --fqbn esp32:esp32:ttgo-lora32-v1 sample`

```Build options changed, rebuilding all
Traceback (most recent call last):
  File "/root/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial
Error: build failed: exit status 1
Error during build```

### Expected behavior

It to compile to a `.bin` file without needing to contact the board.

### Environment

- CLI version `arduino-cli Version: 0.3.7-alpha.preview Commit: 69d86a4 BuildDate: 2019-10-25 16:19:14.377883065 +0000 UTC`
- OS and platform: Ubuntu 18.04.3 LTS

### Additional context

ESP8266 boards build just fine.

In my case, I have a custom Docker image that is set up to build Arduino binaries for several apps across several device types, and offer them over OTA to ESP8266 and RSP32 devices.  When adding the ESP32 build, it seems to error out, but I'm not exactly sure why.

Note that `apt install python-serial` does fix this, and allow the build to complete.

cmaglie, 2019-10-29:
@skandragon `esptool` requires pyserial, even if the serial port is not used at all (I guess that the library is loaded anyway). I think this issue can't be fixed on the `arduino-cli` side and the solution is to install `python-serial` on the test environment.

If you agree I'll close this one as wontfix.

matthijskooijman, 2019-10-29:
Sounds like something to report at esptool, since it should certainly be possible to only load pyserial when actually needed, but that requires changes in esptool obviously.

cmaglie, 2019-11-08:
Issue reported upstream: https://github.com/espressif/esptool/issues/477

closed 2019-11-08 by cmaglie

Sajith95, 2020-02-23:
This cannot be resolved using `python-serial` or `esptool`