arduino/arduino-cli #458: Get error "can't open device /dev/ttyAMC0: No such file" but IDE works

QuinnFreedman, 2019-10-28:
## Bug Report

### Current behavior

I ran:
```
arduino-cli compile --fqbn adafruit:avr:feather32u4 . && \
  arduino-cli upload -b adafruit:avr:feather32u4 -p /dev/ttyAMC0  -t .
```
An `.ino` file in the same directory has a simple 5-line LED blink program.

Compilation works fine but when I run `arduino-cli upload`, I get the message 
```
No new serial port detected.
avrdude: ser_open(): can't open device "/dev/ttyAMC0": No such file or directory
Error during Upload: uploading error: exit status 1
```
I get the same result when I try `AMC1` or `ttyS0`.

When I run `arduino-cli board list`, I get

```
Port         Type              Board Name FQBN Core
/dev/ttyACM0 Serial Port (USB) Unknown
/dev/ttyS0   Serial Port       Unknown
```

But when I run the Arduino IDE it detects the Adafruit Feather 32u4 board (the only one I have plugged in) on `/dev/AMC0` (and is able to upload to it without trouble).

### Expected behavior
I was expecting the CLI to have the same behavior as the IDE, i.e.:
1. Correctly detect and identify the board on whatever port it is on (in `arduino-cli board list`)
2. (much more importantly) Upload the sketch to the device without errors.

### Environment

- CLI version (output of `arduino-cli version`): 
```
arduino-cli Version: 0.6.0 Commit: 3a08b07
```
- OS and platform:
```
$neofetch --stdout

OS: Ubuntu 19.10 x86_64
Host: MS-7C37 1.0
Kernel: 5.3.0-19-generic
Uptime: 2 hours, 15 mins
Packages: 2309 (dpkg), 11 (snap)
Shell: zsh 5.7.1
Resolution: 1920x1080, 1920x1080
DE: GNOME 3.34.1
WM: i3
Theme: SolArc [GTK2/3]
Icons: Arc [GTK2/3]
Terminal: alacritty
Terminal Font: "Hack Nerd Font"
CPU: AMD Ryzen 7 3800X 8- (16) @ 4.249GHz
GPU: AMD ATI Navi 10
Memory: 3694MiB / 32126MiB
```

### Additional context
I initially had some trouble getting the IDE to upload via USB as well. I kept getting various error messages about "permission denied" to write to `/dev/ttyAMC0`. I followed [this answer](https://adafruit.github.io/arduino-board-index/package_adafruit_index.json) and it cleared the problem up for the IDE, but I still get the "does not exist" error on CLI. 

When I try to upload from the IDE, it doesn't actually always work. The board seems to bounce back and forth between `/dev/ttyACM0` and `/dev/ttyACM1` and sometimes uploading from the IDE just hangs with no error message. When it does work, i get this red text in the log:
```
    Software Version = 
.
; No Hardware Version given.

avrdude: error: programmer did not respond to command: set addr
```

But it seems to work fine and the LED on my board changes its blinking with the new code.

mightycandle, 2020-02-21:
If you use ubuntu, go to the terminal and type `sudo chmod --the error link-- 777` that should clear the issue

per1234, 2022-03-31:
The error is caused by a transposition error in the command:

```text
arduino-cli upload -b adafruit:avr:feather32u4 -p /dev/ttyAMC0
```
(Note the "MC" in `/dev/ttyAMC0`)

vs the actual port name:

```text
Port         Type              Board Name FQBN Core
/dev/ttyACM0 Serial Port (USB) Unknown
```

(Note the "CM" in `/dev/ttyAMC0`)

closed 2022-03-31 by per1234