arduino/arduino-cli #474: panic: runtime error: invalid memory address or nil pointer dereference

gdsports, 2019-11-08:
Using IDE 1.8.9 on Ubuntu Linux 18.04, creating a symbolic link in the sketchbook directory to the sketch code works fine. Using a symbolic link is useful when compiling the same sketch using different versions of the IDE.

Using IDE 1.8.10 on same Linux configuration as above, the compile fails like this.

```
/tmp/autobuild_10493/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /tmp/autobuild_10493/arduino-1.8.10/hardware -hardware /tmp/autobuild_10493/arduino-1.8.10/portable/packages -tools /tmp/autobuild_10493/arduino-1.8.10/tools-builder -tools /tmp/autobuild_10493/arduino-1.8.10/hardware/tools/avr -tools /tmp/autobuild_10493/arduino-1.8.10/portable/packages -built-in-libraries /tmp/autobuild_10493/arduino-1.8.10/libraries -libraries /tmp/autobuild_10493/arduino-1.8.10/portable/sketchbook/libraries -fqbn=adafruit:nrf52:feather52840:softdevice=s140v6,debug=l0 -ide-version=10810 -build-path /tmp/arduino_build_371849 -warnings=none -build-cache /tmp/arduino_cache_20114 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.nrfjprog.path=/tmp/autobuild_10493/arduino-1.8.10/portable/packages/adafruit/tools/nrfjprog/9.4.0 -prefs=runtime.tools.nrfjprog-9.4.0.path=/tmp/autobuild_10493/arduino-1.8.10/portable/packages/adafruit/tools/nrfjprog/9.4.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/tmp/autobuild_10493/arduino-1.8.10/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/tmp/autobuild_10493/arduino-1.8.10/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -verbose /tmp/autobuild_10493/arduino-1.8.10/portable/sketchbook/USBMSEBLE/USBMSEBLE.ino
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x833cfe]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xfa4300, 0xc000148800, 0xfa4300, 0x40c3c8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_setup.go:77 +0x45e
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000148800, 0xc0001abe00, 0x2, 0x2, 0x42b001, 0xc000000008, 0xc00009fdf8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc00009fe70, 0xc000148800, 0xfa41f0, 0xc0000a58d0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:198 +0xcc
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:231
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:388 +0x8b1
/tmp/autobuild_10493/arduino-1.8.10/arduino-builder returned 2

Error compiling for board Adafruit Bluefruit Feather nRF52840 Express.
```

The compile succeeds if the sketch directory is copied into the sketchbook directory.

facchinm, 2019-11-08:
Hi @gdsports ,
can you try IDE hourly build? We patched the builder to overcome some other issues and hopefully it solves also this one. Thanks!

gdsports, 2019-11-08:
@facchinm I tried the hourly build but fails the same way.

```
Arduino: 1.8.11 Hourly Build 2019/10/29 10:25 (Linux), Board: "Adafruit Bluefruit Feather nRF52840 Express, 0.2.13 SoftDevice s140 6.1.1, Level 0 (Release)"

/tmp/arduino-nightly/arduino-builder -dump-prefs -logger=machine -hardware /tmp/arduino-nightly/hardware -hardware /tmp/arduino-nightly/portable/packages -tools /tmp/arduino-nightly/tools-builder -tools /tmp/arduino-nightly/hardware/tools/avr -tools /tmp/arduino-nightly/portable/packages -built-in-libraries /tmp/arduino-nightly/libraries -libraries /tmp/arduino-nightly/portable/sketchbook/libraries -fqbn=adafruit:nrf52:feather52840:softdevice=s140v6,debug=l0 -ide-version=10811 -build-path /tmp/arduino_build_388641 -warnings=default -build-cache /tmp/arduino_cache_955824 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/tmp/arduino-nightly/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/tmp/arduino-nightly/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.nrfjprog.path=/tmp/arduino-nightly/portable/packages/adafruit/tools/nrfjprog/9.4.0 -prefs=runtime.tools.nrfjprog-9.4.0.path=/tmp/arduino-nightly/portable/packages/adafruit/tools/nrfjprog/9.4.0 -verbose /tmp/arduino-nightly/portable/sketchbook/USBMSEBLE/USBMSEBLE.ino
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x833c0e]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xfa7360, 0xc0001bae00, 0xfa7360, 0x40c3c8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_setup.go:78 +0x68e
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc0001bae00, 0xc0001c5e00, 0x2, 0x2, 0xc0001c9d01, 0x7ae17b, 0xc0000f4690)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:209 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc0001c9e70, 0xc0001bae00, 0xfa7250, 0xc0000d1ea0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:197 +0xc6
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:230
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:389 +0x837
/tmp/arduino-nightly/arduino-builder returned 2

Error compiling for board Adafruit Bluefruit Feather nRF52840 Express.
```

gdsports, 2019-11-09:
The command line arduino-cli handles symbolic links to the sketch directory.

```
$ ls -ld My*
lrwxrwxrwx 1 me me   18 Nov  8 19:11 MyFirstSketch -> MyFirstSketch.real
drwxr-xr-x 2 me me 4096 Nov  8 19:10 MyFirstSketch.real
$ arduino-cli compile --fqbn arduino:samd:mkr1000 MyFirstSketch
Sketch uses 11048 bytes (4%) of program storage space. Maximum is 262144 bytes.
```

cmaglie, 2019-11-12:
@gdsports we have updated the `arduino-builder` in the nightly builds, would you like to give it another try now? (re-download the nightly build of the IDE)

gdsports, 2019-11-14:
The hourly build "11 November 2019 6:35:7 GMT" works fine with symlinks! This is can be closed if desired.

masci, 2019-11-18:
Closing, thanks!

closed 2019-11-18 by masci