arduino/arduino-cli #480: Recipes for configuring size / memory usage

PaulStoffregen, 2019-11-11:
Please consider making the memory usage summary more configurable with recipes in platform.txt.

Currently arduino-builder assumes there can be only 2 memory regions.  The 2 names are hard-coded.  The 2 maximum sizes come from prefs in boards.txt.

Some hardware has 3 or more distinct memory regions.  I am current struggling with this problem with the latest Teensy board, where combining 2 regions together to fit into a single line causes a lot of user confusion when they try to use memory that Arduino's size summary leads them to believe should be available, but actually isn't... but appears to be free, because of the necessity of shoehorning multiple regions into builder's limited size summary.

Hopefully arduino-builder could someday recognize a group of rules like this:

    recipe.size.name.mymemory="External RAM"
    recipe.size.regex.mymemory=^(?:\.bss\.extram|\.data\.extram)
    recipe.size.maximum.mymemory=131072

This should cause builder to print line of text like this:

> External RAM uses 24412 bytes (19%).  Maximum is 131072 bytes.

Or if a 4th rule gives a name for the use of leftover space...

    recipe.size.name.mymemory="External RAM chip"
    recipe.size.leftover.mymemory="special buffers"
    recipe.size.regex.mymemory=^(?:\.bss\.extram|\.data\.extram)
    recipe.size.maximum.mymemory=131072

Hopefully this 4 rule case could cause line like this to print:

> External RAM uses 24412 bytes (19%), leaving 106660 bytes for special buffers.  Maximum is 131072 bytes.

Alternately, if this is too much trouble, perhaps a rule could be added to simply run a program which will generate the size summery and builder would skip printing its limited-to-2 regions summary?

PaulStoffregen, 2021-02-20:
@facchinm - Any chance I could talk you into adding a raw unfiltered size recipe?  If a platform defines "recipe.rawsize.pattern", just run the program and use whatever it prints to stdout as-is.  No regex matching, no size checks, just copy its message to the output for a user to see.  And of course if the command returns a non-zero status, consider than a fail as if the normal size check fails.

I tried to accomplish this is a postbuild hook.

    recipe.hooks.postbuild.1.pattern="echo" "just show this as the size message, without regex processing"

But the problem is the user never sees this message in non-verbose mode.

For now, I *really* need a way to show more detailed size info than the 2 regex approach can handle.  I really don't want to add yet another patch to the Java IDE, but I just don't see a way to replace the size process that works in all modes.

cmaglie, 2021-03-04:
@PaulStoffregen I've drafted an implementation, let me polish it a bit so I can open a PR and we'll further discuss it there.

closed 2022-02-01 by cmaglie