arduino/arduino-cli #482: arduino-cli broken after ESP32 core file goes missing

ipsod, 2019-11-12:
## Bug Report

### Current behavior
I had this error come up up when running `arduino-cli` after I'd opened Arduino IDE's board manager and it failed to find anything at one `Additional Board Manager URL`:

    Error: loading json index file /home/zero/.arduino15/package\_esp32\_index.json: open /home/zero/.arduino15/package\_esp32\_index.json: no such file or directory
    Error loading index
    Launch 'arduino-cli core update-index' to fix or download indexes.

So, I did as it said:

    $ arduino-cli core update-index
    Updating index: package_index.json downloaded
    Updating index: package_index.json downloaded
    Updating index: package_STM32duino_index.json downloaded
    Error: downloading index https://dl.espressif.com/dl/package_esp32_index.json: Get https://dl.espressif.com/dl/package_esp32_index.json: net/http: TLS handshake timeout
    Error updating index

I tried fixing it by removing the ESP32 board support from within the Arduino IDE's `file->preferences` menu, then updating the board index from within Arduino, but running the above command still gave the same output.  I couldn't do anything with arduino-cli until I restored a backup of my `.arduino15` folder from yesterday.

### Expected behavior
Not failing like that.

### Environment
arduino-cli Version: 0.3.7-alpha.preview Commit: 69d86a4 BuildDate: 2010 UTC

- OS and platform: Linux Mint 19.1

### Additional context
I'm still not able to update the ESP32 core - any idea what's going on there?

cmaglie, 2019-11-13:
> Error: downloading index https://dl.espressif.com/dl/package_esp32_index.json: Get https://dl.espressif.com/dl/package_esp32_index.json: net/http: TLS handshake timeout

this looks like a temporary problem on the download server dl.espressif.com, did it work now? may you try also with the latest version of the CLI?

masci, 2019-12-04:
Closing as it seems a transient problem and it works now. Feel free to reopen should this happen again.

closed 2019-12-04 by masci