arduino/arduino-cli #494: Problem with esp8266, esp32 and digistump installation in arduino-cli

oholen, 2019-11-24:
## Bug Report

### Current behavior
My arduino-cli.yaml
------------------------------------------------------------------------------------------
```
proxy_type: auto
sketchbook_path: /home/jan/Arduino
arduino_data: /home/jan/.arduino15
board_manager:
        additional_urls:
                - http://arduino.esp8266.com/stable/package_esp8266com_index.json
                - http://digistump.com/package_digistump_index.json
                - https://dl.espressif.com/dl/package_esp32_index.json
```
------------------------------------------------------------------------------------------

[jan@s61.linuxpl.com:: ~/.arduino15 ]:$ arduino-cli core update-index
_Updating index: package_index.json downloaded                                                                                                                                                         
Updating index: package_index.json downloaded_

[jan@s61.linuxpl.com:: ~/.arduino15 ]:$ arduino-cli core install esp8266:esp8266
_**Error during install: finding platform dependencies: package esp8266 not found**_

Esp32 and digistump this same.

### Expected behavior
Arduino:sam install is OK.
------------------------------------------------------------------------------------------
[jan@s61.linuxpl.com:: ~/Arduino ]:$ arduino-cli core install arduino:sam
_Downloading packages...
arduino:arm-none-eabi-gcc@4.8.3-2014q1 downloaded                                                                                                                                                     
arduino:bossac@1.6.1-arduino downloaded                                                                                                                                                               
arduino:sam@1.6.12 downloaded                                                                                                                                                                         
Installing arduino:arm-none-eabi-gcc@4.8.3-2014q1...
arduino:arm-none-eabi-gcc@4.8.3-2014q1 installed
Installing arduino:bossac@1.6.1-arduino...
arduino:bossac@1.6.1-arduino installed
Installing arduino:sam@1.6.12...
arduino:sam@1.6.12 installed_
------------------------------------------------------------------------------------------
### Environment

My arduino-cli
------------------------------------------------------------------------------------------
[jan@s61.linuxpl.com:: ~/.arduino15 ]:$ arduino-cli version
_arduino-cli Version: 0.6.0 Commit: 3a08b07_
------------------------------------------------------------------------------------------

My Linux (uname -a)
------------------------------------------------------------------------------------------
Linux s61.linuxpl.com 4.4.41-grsec #1 SMP Tue Jan 10 10:26:33 CET 2017 x86_64 GNU/Linux
------------------------------------------------------------------------------------------
### Additional context
It's been the same for several days.

masci, 2019-11-26:
Can you run the command with `-v`:
```
[jan@s61.linuxpl.com:: ~/.arduino15 ]:$ arduino-cli core update-index -v
```
and post the output?

oholen, 2019-12-04:
`[jan@s61.linuxpl.com:: ~ ]:$ arduino-cli core update-index -v
`

> INFO[0000] Checking for config file in: /home/jan/.arduino15/arduino-cli.yaml 
> INFO[0000] Reading configuration from /home/jan/.arduino15/arduino-cli.yaml 
> WARN[0000] Could not read configuration from /home/jan/.arduino15/arduino-cli.yaml  error="yaml: line 5: found character that cannot start any token"

...

Ok, problem resolved.
My _arduino-cli.yaml_ file is invalid.
I added too many "tabs".
The correct file is below:

```
proxy_type: auto
sketchbook_path: /home/januszc/Arduino
arduino_data: /home/januszc/.arduino15
board_manager:
[tab]additional_urls:
[tab]- http://arduino.esp8266.com/stable/package_esp8266com_index.json
[tab]- http://digistump.com/package_digistump_index.json
[tab]- https://dl.espressif.com/dl/package_esp32_index.json
```

The `[tab]` key should be used only once before addresses.

Thank you **masci** for your mail and hint: `-v`

Now everything works.

masci, 2019-12-04:
I'm glad you fixed the issue!
I'm closing as resolved, feel free to reopen should you find something else.

closed 2019-12-04 by masci