arduino/arduino-cli #503: Compile still unable to compile sketch with version 0.6.0

NormanDunbar, 2019-11-30:
## Bug Report
`Sketch new <name>` no longer creates the new sketch within the sketchbook location in the config. `Compile` does now find the sketches located in sketchbook.

Similar to #406, but the other way around!

### Current behavior
````
# Get config
arduino-cli config dump

proxy_type: auto
sketchbook_path: /home/norman/Arduino
arduino_data: /home/norman/.arduino15
board_manager: {}

# Where am I?
pwd
/home/norman

# Create sketch - outside of sketchbook location
arduino-cli sketch new testingAgain
Sketch created in: /home/norman/testingAgain

# Compile sketch
arduino-cli compile -b arduino:avr:mega testingAgain
Sketch uses 656 bytes (0%) of program storage space. Maximum is 253952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.

# Move sketch to sketchbook
mv testingAgain/ /home/norman/Arduino/
cd /home/norman/Arduino
ls
... testingAgain ...

# Re-compile sketch
arduino-cli compile -b arduino:avr:mega testingAgain
Sketch uses 656 bytes (0%) of program storage space. Maximum is 253952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.
````

So, #406 appears to have been fixed, but where the sketch is created appears to have regressed. Previous versions to 0.6.0, created new sketches within the sketchbook.

Sketches are no longer created within the sketchbook location according to the config. However, they can now be compiled if they are within the current folder or within the sketchbook

### Expected behavior
I expect the `sketch new` to continue to create sketches within the sketchbook area as defined in the config.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.6.0 Commit: 3a08b07
- OS and platform: Linux Mint 19.2 64 bit.

### Additional context
#406 was raised as the sketches were created in sketchbook, but compiled from the current directory. Now it's the other way around!

Thanks for all the hard work you do.

Cheers,
Norm.

masci, 2019-12-04:
Hi @NormanDunbar and thanks for the detailed feedback.

I think the confusion stems from the fact we're trying to get rid of the Sketchbook concept in the CLI in favor of defaulting to "current working dir" in absence of full paths, see https://github.com/arduino/arduino-cli/pull/396 for context.

I'll try to make more explicit this change in the README.

NormanDunbar, 2019-12-04:
Good Evening @masci,

Ok, that makes sense, however, given that the `arduino-cli config dump` command lists the default location as the sketchbook, might I suggest removing that config option as it appears to be redundant in view of #396.

Many thanks.

Cheers,
Norm.

masci, 2019-12-05:
@NormanDunbar yeah that makes sense indeed, specially because we're about to merge https://github.com/arduino/arduino-cli/pull/487 - the config file will change a bit and we can take the chance to remove the sketchbook reference. I'll review all the places where we mention the Sketchbook and fix accordingly, I'll keep this issue open to track this work.

Thanks for the suggestion!

NormanDunbar, 2019-12-05:
Hi @masci. I'm happy with that and qill close this issue.

As ever, thanks for all you all do for us users.

Cheers,
Norm.

closed 2019-12-05 by NormanDunbar