arduino/arduino-cli #505: `board list` does not list virtual ports

kittaakos, 2019-12-03:
## Bug Report
I have copied the bug report from [here](https://github.com/arduino/arduino-pro-ide/issues/119#issuecomment-559412167):

I set up virtual ports on Windows with [`com0com`](http://com0com.sourceforge.net) and mapped `COM5` to `ARDUINO`:
[image]

I can confirm, I can see the ports in the Java IDE:
[image]

And I can also confirm when I run the `board list` on the CLI, I cannot see the virtual ports. I think it is a CLI issue. What do you think @masci?

```
$ ./arduino-cli.exe board list
Port Type        Board Name FQBN Core
COM1 Serial Port Unknown
COM5 Serial Port Unknown
```

```
$ ./arduino-cli.exe board list --format json
[
  {
    "address": "COM1",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "COM5",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  }
]

```

### Current behavior
The CLI does not return with the virtual ports.

### Expected behavior

I can see the exact same port with the CLI as with the Java IDE.

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

### Additional context

masci, 2019-12-04:
This is a known missing feature of the underlying go-serial library, marking as feature request with no clear priority.

per1234, 2021-02-18:
I believe there is now an upstream feature request for this: https://github.com/bugst/go-serial/issues/85