arduino/arduino-cli #508: How to set default processor?

ilgiznurgaliev, 2019-12-04:
Hi.

Could please someone tell me know how to set default processor for specific board.
For example as I can see it is "cpu=atmega328" for Arduino Nano:

```
$ arduino-cli board details arduino:avr:nano
Board name:     Arduino Nano                 

Required tools: arduino:avr-gcc                 7.3.0-atmel3.6.1-arduino5
                arduino:avrdude                 6.3.0-arduino17          
                arduino:arduinoOTA              1.3.0                    

Option:         Processor                       cpu                      
                ATmega328P                    ✔ cpu=atmega328            
                ATmega328P (Old Bootloader)     cpu=atmega328old         
                ATmega168                       cpu=atmega168
```

I would like to set the default to "cpu=atmega328old".

Thanks in advance.

per1234, 2019-12-17:
The default option is the one that comes first in boards.txt:
https://github.com/arduino/ArduinoCore-avr/blob/1.8.2/boards.txt#L143

However, you can use `arduino-cli board attach` to set the default for a particular sketch. Let's say you had a sketch named Foo. This command:
```
arduino-cli board attach arduino:avr:nano:cpu=atmega328old Foo
```
will cause a metadata file named sketch.json to be written to the sketch folder. After that, all operations on that sketch will use the atmega328old option, unless another one is specified in the command's `fqbn` option.

ilgiznurgaliev, 2019-12-18:
Hi. The "board attach" option is exactly what I was needed. Thank you so much for your help!

masci, 2020-01-02:
Closing as resolved, feel free to reopen should you have any other issue on this topic.

closed 2020-01-02 by masci

pouriap, 2022-02-10:
This only works with one options. If you specify more than one option you get an error:
```
arduino board attach -b MiniCore:avr:8:bootloader=no_bootloader:BOD=4v0:LTO=Os_flto:clock=8MHz_internal
```

```
Attach board error: Invalid Device URL format: parse "serial://MiniCore:avr:8:bootloader=no_bootloader:BOD=4v0:LTO=Os_flto:clock=8MHz_internal": invalid port ":clock=8MHz_internal" after host
```