arduino/arduino-cli #510: error getting port list from serial-discovery: read |0: The handle is invalid

kittaakos, 2019-12-10:
## Bug Report
This happened on Windows with the Pro IDE. This is all I can see in the logs:
```
root ERROR Uncaught Exception:  Error: 2 UNKNOWN: error getting port list from serial-discovery: read |0: The handle is invalid.
root ERROR Error: 2 UNKNOWN: error getting port list from serial-discovery: read |0: The handle is invalid.
    at Object.callErrorFromStatus (c:\Users\kittaakos\Desktop\Arduino Pro IDE\resources\app\node_modules\@grpc\grpc-js\build\src\call.js:30:26)
    at Http2CallStream.call.on (c:\Users\kittaakos\Desktop\Arduino Pro IDE\resources\app\node_modules\@grpc\grpc-js\build\src\client.js:96:33)
    at Http2CallStream.emit (events.js:187:15)
    at process.nextTick (c:\Users\kittaakos\Desktop\Arduino Pro IDE\resources\app\node_modules\@grpc\grpc-js\build\src\call-stream.js:79:22)
    at process._tickCallback (internal/process/next_tick.js:61:11)
```

It comes from the CLI when listing the available ports, but I do not know how to consistently reproduce the defect:
https://github.com/arduino/arduino-cli/blob/2049a7afd25bd3c0d70dcf71ca8cded855522187/commands/board/list.go#L116

### Current behavior

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): version: 0.6.0 commit: 2049a7a
- OS and platform: Windows 10 (running in VMWare Fusion `11.5.1 (15018442)` on macOS)

### Additional context

kittaakos, 2019-12-10:
Note: I have been using this CLI version for the last week or so on macOS, and haven't hit this error.

kittaakos, 2019-12-10:
Got another kind of error on Windows, I do not know if it is related to the original one:
```
root ERROR Uncaught Exception:  Error: 2 UNKNOWN: error getting port list from serial-discovery: decoding LIST command:
timeout
root ERROR Error: 2 UNKNOWN: error getting port list from serial-discovery: decoding LIST command: timeout
    at Object.callErrorFromStatus (C:\Users\kittaakos\dev\arduino-editor\node_modules\@grpc\grpc-js\build\src\call.js:30
:26)
    at Http2CallStream.call.on (C:\Users\kittaakos\dev\arduino-editor\node_modules\@grpc\grpc-js\build\src\client.js:96:
33)
    at Http2CallStream.emit (events.js:203:15)
    at process.nextTick (C:\Users\kittaakos\dev\arduino-editor\node_modules\@grpc\grpc-js\build\src\call-stream.js:79:22
)
    at process._tickCallback (internal/process/next_tick.js:61:11)

```

umbynos, 2022-10-12:
The serial discovery system is changed a lot, now we have the pluggable discovery and it seems that this issue is not valid anymore.

closed 2022-10-12 by umbynos