arduino/arduino-cli #518: Doesn't work on non-FHS systems

balsoft, 2019-12-14:
## Bug Report

### Current behavior

```
$ arduino-cli compile --fqbn arduino:avr:nano AnalogSensors/AnalogSensors.ino 
Error during build: build failed: fork/exec /home/balsoft/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++: no such file or directory
```

The build fails because the toolchain (downloaded to `~/.arduino15/packages`) has `/lib/ld-linux-x86-64.so.2` set as interpreter. While it might work on FHS systems, it doesn't work on mine. I can solve this by patching the executables (`for i in $(find /home/balsoft/.arduino15 -t file); do patchelf --set-interpreter /nix/store/xlxiw4rnxx2dksa91fizjzf7jb5nqghc-glibc-2.27/lib/ld-linux-x86-64.so.2 $i; done`) but it is a flimsy solution and won't work long-term (it'll break after the next glibc update and garbage collection). 

### Expected behavior

It works out of the box, independent of the location of `ld-linux-x86-64.so.2`. This can be achieved by building the toolchain statically.

### Environment

- CLI version (output of `arduino-cli version`): `0.3.7-alpha.preview Commit:` (in reality it's built from master branch) The expression is here: https://github.com/balsoft/nur-packages/tree/master/pkgs/arduino-cli
- OS and platform: NixOS GNU/Linux 20.03

### Additional context

I'm not sure this should be reported here. If it isn't, please close the issue and redirect me to the correct place.

facchinm, 2019-12-16:
Mmmh the toolchain is being built as statically as possible (for glibc), we could try compiling against `musl` in the future to solve this but it's not very high priority :slightly_smiling_face:

balsoft, 2019-12-16:
Ah, actually that can be fixed by removing the interpreter from the files, static linking has nothing to do with my actual problem.  Maybe you can do that? I believe it should work on any modern Linux distro that supports binfmt.

balsoft, 2019-12-16:
To fix my problem, we need to link the toolchain with `--no-dynamic-linker` which should work on all platforms that support binfmt (so all of them).

ericonr, 2020-06-06:
Hi there! Do you think it would be possible instead to build musl variants and have arduino-cli auto detect the platform? Void Linux ships arduino-cli in its musl variant, but has to resort to `gcompat` in order to run the AVR compiler, and quite a few more hacks to run the SAM one. I recently dealt with this here https://github.com/void-linux/void-packages/issues/22616#issuecomment-639968663

Where can I find the build configuration, if I were to try and help with it?

balsoft, 2020-06-06:
@ericonr  https://github.com/balsoft/nur-packages/blob/master/pkgs/arduino-cli/default.nix is the package description. `deps.nix` is generated from `go.sum` or `go.mod`, don't remember which one.

per1234, 2022-03-31:
Someone with write permission in [the `arduino/toolchain-avr` repo](https://github.com/arduino/toolchain-avr) (@facchinm, @cmaglie, ...), please transfer the issue over there. The issue doesn't seem to be anything that can be resolved by changing the Arduino CLI code base.