arduino/arduino-cli #519: Portable platforms

caxapexac, 2019-12-15:
I need to have cli to run it without internet. I need nano platform configuration and store it with the executable folder. How to do it?

UnDetected, 2020-01-31:
How use Arduino-CLI in portable mode?

UnDetected, 2020-01-31:
In Windows use next command

`arduino-cli config init`

After this, you will get a simple config file. Inside the arduino-cli.yaml exists section **directories**. Just set relative path to needed directories and you get a portable version.

umbynos, 2020-02-03:
Using `arduino-cli config -h` will show you the available commands usable to alter the config.
With `arduino-cli config dump` you can see what's the actual config.
As @UnDetected said you can create a new config and change the paths under section **directories** where you prefer. This will allow you to make a portable installation

umbynos, 2020-02-03:
1. `curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/cli_portable sh` *cli_portable* is the folder in which the cli will be installed. It clould be also in a flash drive
2. `cd cli_portable`
3.  `./arduino-cli config init` will initialize the config file and will tell you where will be located that file. Just move that file in your `cli_portable` folder.
4. Edit the `arduino-cli.yaml` modifying section *directories* (in particular *data* and *downloads*)
5. :heavy_check_mark:

closed 2020-04-14 by github-actions[bot]