arduino/arduino-cli #520: macos Catalina  not

Topppy, 2019-12-17:
## Bug Report

### Current behavior

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

### Additional context

a7hybnj2, 2019-12-19:
This affects me too.

```
arduino-cli compile --fqbn arduino:avr:uno ./
Build options changed, rebuilding all
Error: fork/exec /Users//Library/arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags: bad CPU type in executable
Compilation failed.
make: *** [all] Error 1
```

rsora, 2020-01-20:
Hi @Topppy, thanks for your bug report! 
The tool you mentioned was rebuilt and updated in the CLI list of additional tools to download in this PR: https://github.com/arduino/arduino-cli/issues/520

Could you please update your CLI version to the latest (0.7.2) and try again?

closed 2020-03-13 by github-actions[bot]

Topppy, 2020-06-29:
> Hi @Topppy, thanks for your bug report!
> The tool you mentioned was rebuilt and updated in the CLI list of additional tools to download in this PR: #520
> 
> Could you please update your CLI version to the latest (0.7.2) and try again?

Thanks for your work.