arduino/arduino-cli #522: Additional source files that start with RCS are no longer included in a project

tobiasisenberg, 2019-12-17:
## Bug Report

### Current behavior

[MyOwnTest.zip](https://github.com/arduino/arduino-cli/files/3974048/MyOwnTest.zip)
If you try to compile the attached example that includes external local files, this fails. This used to work in 1.8.9. The effect started in 1.8.10. This only happens for specific included file names, namely *.h and *.cpp files that start with "RCS".

Interestingly, if the name of *.ino file is "test" or "bug" I get a completely different compile error:
`panic: runtime error: invalid memory address or nil pointer dereference`
Very strange, you can try it out by renaming *.ino and directory to bug.

### Expected behavior

The code compiles normally.

### Environment

- CLI version (output of `arduino-cli version`): 1.8.10
- OS and platform: Windows 10, 64bit

### Additional context

matthijskooijman, 2019-12-17:
I suspect this changed due to https://github.com/arduino/arduino-cli/pull/366

masci, 2019-12-18:
That function is supposed to filter out files within folders called `RCS` so this is a bug, a source tree like the attached archive must compile without problems.

PR #537 (2020-01-09): Allow naming sketches like "RCS" and "CVS"
Fixes #522 

The filter we use to ignore files contained in version control systems' cache was applied to the sketch folder itself, thus preventing a sketch to be named like `RCS`.

This PR increases resiliency by avoiding panics if a sketch name couldn't be found and applies blacklist filters only to folders contained in the sketch, i.e. excluding the root folder.
closed 2020-01-09 by masci