arduino/arduino-cli #553: gRPC Init command with custom directories not working

andmeyee, 2020-01-11:
I am playing around with arduino-cli's gRPC interface and Python. I am not able to set custom directories for dataDir, sketchbookDir and downloadsDir using the gRPC Init() method.All other gRPC methods seem to work as expected.

## Bug Report

### Current behavior
If I call `client.Init(req)` with req.dataDir set to a custom directory, it is still using the directories configured while starting the daemon.

### Expected behavior
arduino-cli should use the directories specified within the `req` argument

### Environment

- arduino-cli 0.7.2
- Linux (Ubuntu)
- Python 3

### Additional context
None

per1234, 2020-01-11:
Hi @andmeyee there is a new Settings interface that has replaced the system of setting the configuration via `Init`:
https://github.com/arduino/arduino-cli/pull/521

The configuration passed to Init is now ignored:
https://github.com/arduino/arduino-cli/pull/530

andmeyee, 2020-01-11:
Thanks @per1234  for pointing that out. I was not aware of the new settings interface, but using it works as expected. Updating the client_example might be a good idea to make others aware of this.

Using the settings interface, I assume that I change the corresponding settings (e.g. directories) on a global basis for the running arduino-cli daemon. This means that I am not able to set different settings per instance anymore? That would be nice if multiple applications are connecting to the arduino-cli daemon.

masci, 2020-01-23:
Hi @andmeyee you're right, at this moment settings are global, and this is by design since we wanted to make consistent the settings API that previously was ambiguous.

I understand the requirements of being capable to override some settings for specific instances, I'm marking this issue as a Feature Request and will work on an API to achieve that, thanks for your feedback.

andmeyee, 2020-01-23:
Thanks @masci for marking this as a feature request. Happy to give it a try if a first implementation is available.

kittaakos, 2020-02-27:
> Hi @andmeyee you're right, at this moment settings are global,

@masci, before the introduction of the new `Settings` API, two services were available: the `ArduinoCore` and the `Monitor`. I thought the clear separation of these two services was based on the fact that the `Monitor` service is entirely independent of the `ArduinoCore` since one could tune the `datadir`, `sketchbookdir`, `downloadsdir`, and the `boardmanageradditionalurlsList ` in the `InitRequest`'s `Configuration`. So it made sense to have one single gRPC monitor client and several Arduino core clients, as they were initialized differently.

My questions would be the followings:
 - Knowing that the settings are global, would you still recommend downstream consumers of the CLI to maintain multiple `ArduinoCore` gRPC clients, or is it safe to have one and share it?
 - If it is safe to share one `ArduinoCore` gRPC client, what is the benefit of having three different types of clients (`Settings`, `ArduinoCore`, and `Monitor`)? As a consumer of the CLI, I have to maintain the lifecycle of three different gRPC clients, although they're connected to the same server.

Any kind of feedback is welcome on this topic; I am trying to figure out whether we can simplify the architecture of the Pro IDE. Thank you!