arduino/arduino-cli #557: Add flag to print the sketch compiler command

spoenemann, 2020-01-15:
## Feature Request

In order to support language servers for Arduino sketches, it is necessary to know the exact compiler command used by the CLI to compile a sketch. For example, the output of compiling the "KeyboardAndMouseControl" example is

```
Detecting libraries used...
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /var/folders/lx/_zc3wg6d2px3lqdwj8d35jvr0000gn/T/arduino-sketch-647BD920C1206A9070C1F514DE932FA7/sketch/KeyboardAndMouseControl.ino.cpp -o /dev/null
Alternatives for Keyboard.h: [Keyboard@1.0.2]
ResolveLibrary(Keyboard.h)
  -> candidates: [Keyboard@1.0.2]
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/spoenemann/Documents/Arduino/libraries/Keyboard/src /var/folders/lx/_zc3wg6d2px3lqdwj8d35jvr0000gn/T/arduino-sketch-647BD920C1206A9070C1F514DE932FA7/sketch/KeyboardAndMouseControl.ino.cpp -o /dev/null
Alternatives for HID.h: [HID@1.0]
ResolveLibrary(HID.h)
  -> candidates: [HID@1.0]
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/spoenemann/Documents/Arduino/libraries/Keyboard/src -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/HID/src /var/folders/lx/_zc3wg6d2px3lqdwj8d35jvr0000gn/T/arduino-sketch-647BD920C1206A9070C1F514DE932FA7/sketch/KeyboardAndMouseControl.ino.cpp -o /dev/null
Alternatives for Mouse.h: [Mouse@1.0.1]
ResolveLibrary(Mouse.h)
  -> candidates: [Mouse@1.0.1]
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/spoenemann/Documents/Arduino/libraries/Keyboard/src -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/HID/src -I/Users/spoenemann/Documents/Arduino/libraries/Mouse/src /var/folders/lx/_zc3wg6d2px3lqdwj8d35jvr0000gn/T/arduino-sketch-647BD920C1206A9070C1F514DE932FA7/sketch/KeyboardAndMouseControl.ino.cpp -o /dev/null
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/spoenemann/Documents/Arduino/libraries/Keyboard/src -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/HID/src -I/Users/spoenemann/Documents/Arduino/libraries/Mouse/src /Users/spoenemann/Documents/Arduino/libraries/Keyboard/src/Keyboard.cpp -o /dev/null
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/spoenemann/Documents/Arduino/libraries/Keyboard/src -I/Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/HID/src -I/Users/spoenemann/Documents/Arduino/libraries/Mouse/src /Users/spoenemann/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/HID/src/HID.cpp -o /dev/null
/Users/spoenemann/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb  [... cut ...] d-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322] (GNU Tools for ARM Embedded Processors)
```
Here we can find the built-in configuration of the compiler, like default flags.

All this information is essential for providing correct language support in an IDE.

matthijskooijman, 2020-01-16:
But do you then need just the command to compile the sketch? There are different commands for .c, .cpp, .S files, and a command for linking the entire sketch, all of which are printed for each file where appropriate already. Are you suggesting to only print all these, or just the sketch command?

matthijskooijman, 2020-01-16:
Another approach that might be related is to generate a "Compilation database" with all compiler commands that have been run. This seems to be a somewhat standardized file format, see https://github.com/arduino/arduino-builder/issues/272

spoenemann, 2020-01-17:
> Are you suggesting to only print all these, or just the sketch command?

The command that is relevant for providing language support to the sketch file is the one that compiles the sketch source.

But yes, at some point it might be interesting to have the information for other files, too. So generating a JSON file as [specified by the LLVM project](https://clang.llvm.org/docs/JSONCompilationDatabase.html) would be a viable alternative, and would be even more flexible.

umbynos, 2022-10-13:
Closing because we implemented the compilation database functionality

closed 2022-10-13 by umbynos