arduino/arduino-cli #561: Is is possible to change library folder?

anhtu91, 2020-01-18:
Hi everyone,

firstly, thank you guys a lot for the development of arduino-cli. I have used it for my project and it is very very useful. But I still have a question: Is is possible to change library folder from C:\User\{Current login user}\AppData\Local\Arduino15 to another folder (For example: C:\Temp)?

Thanks in advance.

per1234, 2020-01-18:
You can configure the folders used by arduino-cli in the arduino-cli.yaml configuration file.

Create the file with this command:
```shell
arduino-cli config init
```
The output of the command will show you where the file was created. Then you can open that file with a text editor, edit the paths you want to change, and save it.

Note that there are multiple locations you can have an arduino-cli.yaml file. The locations, in order of priority are:
1. Current folder
1. arduino-cli installation folder
1. C:\Users\{Current login user}\AppData\Local\Arduino15

closed 2020-01-18 by per1234

TCB13, 2023-01-24:
It would be nice to have a way to set the library folder with in sketch.yaml!!

TCB13, 2023-01-24:
Possible with:
````
 --libraries strings            List of custom libraries dir paths separated by commas. Or can be used multiple times for multiple libraries dir paths.
````
More info: https://arduino.github.io/arduino-cli/0.20/commands/arduino-cli_compile/