arduino/arduino-cli #566: [Windows] Move user directory away from Documents?

spoenemann, 2020-01-23:
## Bug Report / Discussion

The Pro IDE issue https://github.com/arduino/arduino-pro-ide/issues/179 is caused by a Windows 10 feature called [Controlled Folders](https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/controlled-folders) (see also https://github.com/PowerShell/PowerShell/issues/8857). As a consequence, action is required by users who have this feature active on their Windows installations:

> Controlled folder access works by only allowing apps to access protected folders if the app is included on a list of trusted software. If an app isn't on the list, Controlled folder access will block it from making changes to files inside protected folders.
>
> Apps are added to the trusted list based upon their prevalence and reputation. Apps that are highly prevalent throughout your organization, and that have never displayed any malicious behavior, are deemed trustworthy and automatically added to the list.
>
> Apps can also be manually added to the trusted list via SCCM and Intune. Additional actions, such as adding a file indicator for the app, can be performed from the Security Center Console.
>
> (from the [Microsoft Docs](https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/controlled-folders))

Maybe it would be better to use another user sketch / library directory in the future in order to avoid such hassle, e.g. `$HOME\Arduino` instead of `$HOME\Documents\Arduino`. However, that would also mean to provide migration utility, e.g. by offering to copy / move the existing sketches and libraries to the new location.

If the user directory stays the same, the CLI and Pro IDE documentation and FAQ should explain how to resolve problems related to Controlled Folders.

PrashantDandriyal, 2020-03-04:
The simplified issue is: _Controlled folder access_  sees Arduino as an untrusted application and hinders its access to the working/save directory ```HOME\Documents\Arduino```.
> Controlled folder access is especially useful in helping to protect your documents and information from ransomware that can attempt to encrypt your files and hold them hostage.

From the microsoft docs on [Controlled folder](https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/controlled-folders), only Windows 10 and Windows Server 2019 users suffer from this. 
>Controlled folder access is supported on Windows Server 2019 as well as Windows 10 clients. **It can be turned on via the Windows Security App, or from the Microsoft Endpoint Configuration Manager and Intune, for managed devices**. Controlled folder access works best with Microsoft Defender Advanced Threat Protection, which gives you detailed reporting into controlled folder access events and blocks as part of the usual alert investigation scenarios.

Until the issue is solved by Arduino, simple hacks include

- Adding Arduino IDE to _trusted list_
> Apps can also be manually added to the trusted list via Configuration Manager and Intune. Additional actions, such as adding a file indicator for the app, can be performed from the Security Center Console.
- Whitelist Arduino IDE 
>You can also allow or [whitelist apps](https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/customize-controlled-folders#allow-specific-apps-to-make-changes-to-controlled-folders) to give them access to the protected folders.
- Disabling _Windows Defender Antivirus real-time protection_
>Controlled folder access requires enabling Windows Defender Antivirus real-time protection.
_Note: The above methods have not been tested on the Arduino IDE but quoted from Microsoft's site for general apps_

matthijskooijman, 2020-10-08:
Hm, this sounds annoying.

Moving the (default) sketchbook location outside of the Documents folder (i.e. one level up) does not seem a useful solution, because there it becomes less visible to to users. Sketches *are* documents, just like one could create a new text document and would want to save it in the documents folder.

How does this work for other applications? Is there a mechanism that if you use a standard Windows "Save file..." dialog to select a file that you can bypass this protection? Or is every Windows application in the future going to run into issues saving a user's documents? And will that then result in applications saving to random other locations resulting in a big mess?

I don't have too much right to speak here, not being a Windows user, but this seems like an issue that should maybe solved on the Windows (user) end rather than then by changing defaults on the arduino-cli side?

umbynos, 2022-10-13:
Since this is not affecting many users, I would prefer to have a troubleshooting guide in the docs