arduino/arduino-cli #568: Installation failed on raspberry

klorydryk, 2020-01-25:
## Bug Report

### Current behavior

brew install arduino-cli

### Expected behavior

arduino-cli installation

### Environment

- OS and platform:
Debian 9 on Raspberry v1

### Additional context
```
brew install arduino-cli
/home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/brew.sh: line 4: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/brew.sh: line 4: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
Updating Homebrew...
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/brew.sh: line 4: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
==> Installing dependencies for arduino-cli: go
==> Installing arduino-cli dependency: go
==> Downloading https://dl.google.com/go/go1.13.6.src.tar.gz
Already downloaded: /home/pi/.cache/Homebrew/downloads/eb50aca97a42920b9531b23a11302a07f744adfe5d1e950f8c264c01bb091c94--go1.13.6.src.tar.gz
==> Downloading https://dl.google.com/go/go1.9.linux-armv6l.tar.gz
Already downloaded: /home/pi/.cache/Homebrew/downloads/dc0bab33f6101f4f77840874427d15d7bd84943c7faa489729a5de7d8b0b82b8--go1.9.linux-armv6l.tar.gz
==> ./make.bash --no-clean
Last 15 lines from /home/pi/.cache/Homebrew/Logs/go/01.make.bash:
cmd/internal/obj.Flushplist(0x115ec000, 0x1178bb84, 0x1140cb40)
	/usr/local/go/src/cmd/internal/obj/plist.go:108 +0x6b0
cmd/compile/internal/gc.(*Progs).Flush(0x11798d80)
	/usr/local/go/src/cmd/compile/internal/gc/gsubr.go:87 +0x94
cmd/compile/internal/gc.compileSSA(0x11462a90, 0x0)
	/usr/local/go/src/cmd/compile/internal/gc/pgen.go:244 +0xa4
cmd/compile/internal/gc.compile(0x11462a90)
	/usr/local/go/src/cmd/compile/internal/gc/pgen.go:219 +0x20c
cmd/compile/internal/gc.funccompile(0x11462a90)
	/usr/local/go/src/cmd/compile/internal/gc/dcl.go:1049 +0xa4
cmd/compile/internal/gc.Main(0x6c5f88)
	/usr/local/go/src/cmd/compile/internal/gc/main.go:585 +0x27e4
main.main()
	/usr/local/go/src/cmd/compile/main.go:49 +0x7c
go tool dist: FAILED: /tmp/go-20200125-7484-16m28tr/go/gobootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 2

READ THIS: https://docs.brew.sh/Troubleshooting
```

klorydryk, 2020-01-27:
(it works unsing curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh )

thekunalsaini, 2020-03-03:
@klorydryk use stable version of debain

thekunalsaini, 2020-03-03:
If this is not working let me know

vicpb, 2020-03-12:
@klorydryk just clone the last release. (git clone)

umbynos, 2022-10-13:
Hello @klorydryk, I think that linux arm support for homebrew is problematic:

> Homebrew can run on 32-bit ARM (Raspberry Pi and others) and 64-bit ARM (AArch64), but no binary packages (bottles) are available. Support for ARM is on a best-effort basis. Pull requests are welcome to improve the experience on ARM platforms.

Form https://docs.brew.sh/Homebrew-on-Linux#arm

I suggest to download a release from https://github.com/arduino/arduino-cli/releases/latest

closed 2022-10-13 by umbynos