arduino/arduino-cli #572: Library installed in sketchbook is given priority when core bundled library is better architecture match

per1234, 2020-01-31:
## Bug Report

### Current behavior
1. `./arduino-cli core update-index --additional-urls https://arduino.esp8266.com/stable/package_esp8266com_index.json`
1. `./arduino-cli core install esp8266:esp8266 --additional-urls https://arduino.esp8266.com/stable/package_esp8266com_index.json`
1. `./arduino-cli lib install SD`
1. Create a sketch named Foo that contains the following:
    ```c++
    #include <SD.h>
    void setup(){}
    void loop(){}
    ```
1. `./arduino-cli compile --fqbn esp8266:esp8266:generic Foo`

Compilation fails:
```
C:\Users\per\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
Multiple libraries were found for "SD.h"
 Used: C:\Users\per\Documents\Arduino\libraries\SD
 Not used: C:\Users\per\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SD
```

### Expected behavior
The `architectures` field of C:\Users\per\Documents\Arduino\libraries\SD\library.properties:
https://github.com/arduino-libraries/SD/blob/1.2.4/library.properties#L9
```
architectures=*
```

The `architectures` field of C:\Users\per\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SD
https://github.com/esp8266/Arduino/blob/2.6.3/libraries/SD/library.properties#L9
```
architectures=esp8266
```

Since the library bundled with the core is a better architecture match than the one installed in the sketchbook (explicit match > wildcard match), it should get priority. This is how the priority system has worked historically.

### Environment

- CLI version (output of `arduino-cli version`): 0.7.0-rc1 Commit: afdf259
- OS and platform: Windows 10 64 bit

### Additional context
The bug does not occur at fc9b70bfc5ec51c00e8da3ba02592ed7f1bdb508.

The bug does occur at cca693639b7cd44cf0705036ea8d3212063b0d64.

I remember we had similar problems during the development of arduino-builder. It might be helpful to clearly document the intended library include priorities. This would make it easy to create tests to ensure the correct priorities are maintained.

Originally reported at:
https://forum.arduino.cc/index.php?topic=661342.msg4458460#msg4458460

ubidefeo, 2020-01-31:
Thanks for reporting, @per1234 
I just bumped into this a few hours ago because on a system I still had an old version of ArduinoIoTCloud which was manually installed during development.
I was sure the one installed by `arduino-cli` was up-to-date and it compiled without issues.
Then I saw the lib in the sketch's `libraries` folder, chucked it and it all just worked

ubidefeo, 2020-01-31:
I'd also add that, as a development aid, we should allow a flag in the CLI to permit the use of a llibrary from the user folder to override the core one

jandrassy, 2020-02-03:
ESP32 boards package has this problem with WiFi library. builder uses the old WiFi library bundled with IDE instead of ESP32 WiFi library

jandrassy, 2020-02-03:
The IDE version 1.8.11 should be removed from download page because of this issue. 

https://stackoverflow.com/questions/60036869/multiple-libraries-were-found-for-wire
https://stackoverflow.com/questions/60031945/esp32-how-to-assign-a-string-to-a-char-pointer

commit fe48668a (2020-02-06): Fix libraries priority selection (again) (#574)

* Reorganized tests

* Fixed library.IsArchitectureIndependent method.

Now it returns true for libraries that do not specify "architecture"
fiels.

* Fixed yet another even more convoluted case in bundle lib selection

Fix #572

* Makes linter happy...
closed 2020-02-06 by cmaglie