arduino/arduino-cli #581: Panic runtime error

jgromes, 2020-02-12:
Hi, I'm running into the following issue when trying to compile any Arduino sketch:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x834aba]

goroutine 1 [running]:
github.com/arduino/go-properties-orderedmap.(*Map).Get(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/go-properties-orderedmap/properties.go:210
github.com/arduino/arduino-cli/legacy/builder.(*WipeoutBuildPathIfBuildOptionsChanged).Run(0xf47308, 0x1340e780, 0xf47308, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/wipeout_build_path_if_build_options_changed.go:45 +0x15a
github.com/arduino/arduino-cli/legacy/builder.(*ContainerBuildOptions).Run(0xf47308, 0x1340e780, 0xf47308, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_build_options.go:35 +0x115
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x1340e780, 0x13433e1c, 0x20, 0x20, 0x1, 0x2e604fc, 0xfc)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:195 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x13463f40, 0x1340e780, 0xf47254, 0x13407398)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:102 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:211
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:399 +0x670
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino Uno.
```

I'm on Windows 7 (64 bit). It started happening a few hours ago on 1.8.10. Tried updating to 1.8.11 and the latest hourly build (2020/02/03 07:33) but both produce the same error message. Something to note is that my computer shut down due to power outage while compiling - this issue appeared after that.

Any pointers on what to try to fix this would be greatly appreciated.

thekunalsaini, 2020-03-03:
@cmaglie @jgromes you can check your build folder the error is not in file but you have to do proper setup.
Thank you

mianqi2016, 2020-06-27:
I met the same problem, when I updated Arduino IDE from 1.8.11 to 1.8.13, it was OK.

cmaglie, 2022-10-13:
A lot of time passed since we have no way to reproduce I'll close this as "solved". If there are still problems, I think it's better to open a new issue with updated information to reproduce the error.

closed 2022-10-13 by cmaglie