arduino/arduino-cli #584: Install script doesn't work properly on v0.8.0

SConaway, 2020-02-15:
## Bug Report
Install script doesn't work properly on v0.8.0. I think this might be because the binary is not chmod-ed +x.

````
macos-mojave-10:~ steven$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
Installing in /Users/steven/bin
ARCH=64bit
OS=macOS
Using curl as download tool
TAG=0.8.0
CLI_DIST=arduino-cli_0.8.0_macOS_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.8.0_macOS_64bit.tar.gz
arduino-cli not found. You might want to add /Users/steven/bin to your $PATH
sh: line 185: /Users/steven/bin/arduino-cli: Permission denied
Failed to install arduino-cli
macos-mojave-10:~ steven$ 
````

### Environment

- CLI version (output of `arduino-cli version`): N/A
- OS and platform: macOS 10.14.4 and 10.13.6

zmoog, 2020-02-17:
Hi @SConaway, thank you for taking some time to write this bug report!

The install script is failing due to a missing execution flag on the binary: we are investigating to find the root cause, but in meantime we'll fix the v0.8.0 files in our CDN.

zmoog, 2020-02-17:
@SConaway can you delete the old binary from your machine and try to run the install script again?

SConaway, 2020-02-17:
That fixed it. 

[Failing build](https://travis-ci.com/SConaway/travis-arduino/jobs/287666514)
[Working build](https://travis-ci.com/SConaway/travis-arduino/jobs/288158674)

closed 2020-02-17 by SConaway

zmoog, 2020-02-18:
Great! Thank you for your feedback, happy hacking!