arduino/arduino-cli #588: [proposal] Release the `proto` files with the CLI

kittaakos, 2020-02-19:
## Proposal

### Current behavior

As a gRPC consumer of the CLI, I have to generate the JS/TS modules from the `proto` files. It would be nice to ship the releases with the `proto` files so that one can make sure the CLI is compatible with the gRPC API.

As a possible workaround for this limitation:
 - one can download the CLI,
 - run `./arduino-cli version`,
 - parse the output:
```
./arduino-cli version
arduino-cli Version: 0.8.0 Commit: 64c501d
```
 - clone arduino/arduino-cli and checkout the 64c501d commit,
 - run the  TS/JS generation from the  proto files

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

### Additional context

umbynos, 2022-10-13:
Hi @kittaakos, is this something that is still needed? 
I would not add more complexity in our released archives, maybe a workaround would be to just add a new archive to the releases including only the proto files. But at this point you have the same result as git cloning the repo and checking out the release tag.

kittaakos, 2022-10-17:
> maybe a workaround would be to just add a new archive to the releases including only the proto files

Sounds great.

> But at this point you have the same result as git cloning the repo and checking out the release tag.

How do you expect anybody to develop against the CLI when the APIs are unavailable as part of the release? It's not only for IDE2 but for the public. In fact, IDE2 has to build the CLI quite often from a branch, so distributing the `.proto` files is not a silver-bullet solution.

umbynos, 2022-10-18:
We will make a special archive called `arduino-cli_0.27.1_proto.zip` and add it to the release files. The files that will be added to the archive will be [these ones](https://github.com/arduino/arduino-cli/tree/master/rpc) filtering out the `.go` ones. Would that be good for you?

kittaakos, 2022-10-18:
Fantastic! 😍