arduino/arduino-cli #591: Ability to specify SoftDevice

joachimhs, 2020-02-21:
When compiling for micro:bit, it is necessary to provide SoftDevice, either "none", "S110" og "D130" for the micro:bit.

per1234, 2020-02-21:
You can specify custom menu options via the FQBN. For example, if you wanted to compile for the micro:bit using the **Tools > Softdevice > None** option, the command would look something like this:
```shell
arduino-cli compile --fqbn sandeepmistry:nRF5:BBCmicrobit:softdevice=none Foobar
```
Using the **Tools > Softdevice > S110** option, the command would look something like this:
```shell
arduino-cli compile --fqbn sandeepmistry:nRF5:BBCmicrobit:softdevice=s110 Foobar
```
Using the **Tools > Softdevice > S130** option, the command would look something like this:
```shell
arduino-cli compile --fqbn sandeepmistry:nRF5:BBCmicrobit:softdevice=s130 Foobar
```

The option identifiers are found in boards.txt:
https://github.com/sandeepmistry/arduino-nRF5/blob/0.6.0/boards.txt#L371-L387
```
BBCmicrobit.menu.softdevice.none=None
BBCmicrobit.menu.softdevice.none.softdevice=none
BBCmicrobit.menu.softdevice.none.softdeviceversion=

BBCmicrobit.menu.softdevice.s110=S110
BBCmicrobit.menu.softdevice.s110.softdevice=s110
BBCmicrobit.menu.softdevice.s110.softdeviceversion=8.0.0
BBCmicrobit.menu.softdevice.s110.upload.maximum_size=151552
BBCmicrobit.menu.softdevice.s110.build.extra_flags=-DNRF51 -DS110 -DNRF51_S110
BBCmicrobit.menu.softdevice.s110.build.ldscript=armgcc_s110_nrf51822_xxaa.ld

BBCmicrobit.menu.softdevice.s130=S130
BBCmicrobit.menu.softdevice.s130.softdevice=s130
BBCmicrobit.menu.softdevice.s130.softdeviceversion=2.0.1
BBCmicrobit.menu.softdevice.s130.upload.maximum_size=151552
BBCmicrobit.menu.softdevice.s130.build.extra_flags=-DNRF51 -DS130 -DNRF51_S130
BBCmicrobit.menu.softdevice.s130.build.ldscript=armgcc_s130_nrf51822_xxaa.ld
```

closed 2020-02-21 by per1234

joachimhs, 2020-02-22:
Thank you for the fast response and information!