arduino/arduino-cli #594: Trying to use Arduino-CLI on Banana PI M2 Berry fails

d3v3lop3rDE, 2020-02-23:
I have downloaded the arduino-cli onto my BPI and trying to reconfigure the arduino behind it. When I execute ./arduino-cli board list it gives me the following error:
Error detecting boards: There were errors loading platform indexes

The same thing happen for the attatch command.

ubidefeo, 2020-02-23:
@d3v3lop3rDE 
have you gone through the "How to use it" section?
You need to update your indexes running `arduino-cli core update-index` to let the CLI update the list of available cores/platforms

>The first thing to do upon a fresh install is to update the local cache of available platforms and libraries by running:

>>>
```$ arduino-cli core update-index
Updating index: package_index.json downloaded
```

d3v3lop3rDE, 2020-02-23:
Thank you so much.

closed 2020-02-23 by d3v3lop3rDE

ubidefeo, 2020-02-23:
🙏