arduino/arduino-cli #601: [debug] Support breakpoints on sketch libraries

spoenemann, 2020-03-04:
## Bug Report

See https://github.com/arduino/arduino-pro-ide/issues/217 for a detailed description of the problem.

**Cause of this problem:**
The CLI copies sketch libraries to a temp folder and compiles them from there. With this approach, the source path included in the debugging information is in the temp folder, not the original sketch folder, so breakpoints set on files in the sketch folder are not considered.

**Possible solutions:**
 * Don't copy sketch libraries to the temp folder.
 * Add `#line` compiler directives to the copied files like it's done by the .ino preprocessor. These directives are understood by gcc so it uses the correct source paths in the debugging information.

matthijskooijman, 2020-03-04:
I was under the impression that #line directives are already added? At least the java IDE does so. IIUC the pro IDE uses arduino-cli to compile (whereas the java IDE uses arduino-builder which uses the legacy building code in arduino-cli), so maybe there is a difference between both compilation flows?

Did you check the generated .ino.cpp file to see if there are indeed no line directives in there?

spoenemann, 2020-03-04:
> Did you check the generated .ino.cpp file to see if there are indeed no line directives in there?

The generated .ino.cpp does include the line directives, that's why setting breakpoints in the .ino file works out of the box. But additional .c / .cpp files used by the sketch are copied unchanged.

matthijskooijman, 2020-03-04:
Ah, I totally misread your report, sorry.

Also see https://github.com/arduino/arduino-builder/issues/323 which requests the same thing, and https://github.com/arduino/arduino-builder/pull/325 which implements this in arduino-builder.

Also, I would really like it if we could change the build process to not require these copies at all (so just compile files directly from the sketch directory, rather than printing them), but I think it is tricky to get right (especially wrt getting `#include "foo.h"` directives to work as expected). I think I wrote a bit about this somewhere sometime, but I don't remember where :-p

per1234, 2021-09-30:
Closing as fixed by https://github.com/arduino/arduino-cli/pull/1224

closed 2021-09-30 by per1234