arduino/arduino-cli #620: The name of the board changes after installing the corresponding boards package

kittaakos, 2020-03-13:
## Bug Report

### Current behavior

So the problem is the following; without anything installed, when I list (`core seacrh`) the boards inside the `Arduino AVR Boards` boards package, I can see `Arduino Mega`. I install the boards package and run the same core; there is no `Arduino Mega` anymore. But I can see `Arduino Mega or Mega 2560` instead.

This is critical for the Pro IDE because we cannot set the FQBN of a currently selected board after installing the corresponding boards package. (CC: @cmaglie)

Here are the steps (for macOS, please modify the `rm -rf` command based on the output of `arduino-cli config dump`):
```
rm -rf ~/Documents/Arduino && rm -rf ~/Library/Arduino15 \
&& ./arduino-cli version \
&& ./arduino-cli core update-index \
&& ./arduino-cli core search "Arduino AVR Boards" --format json \
&& ./arduino-cli core install arduino:avr \
&& ./arduino-cli core search "Arduino AVR Boards" --format json
```

Before the boards package installation:
```json
      {
        "name": "Arduino Mega"
      },
```

After the installation:
```json
     {
        "name": "Arduino Mega or Mega 2560",
        "fqbn": "arduino:avr:mega"
      },
```

### Expected behavior

The name of the boards do not change.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.9.0 Commit: e4d02b4`
- OS and platform: macOS 10.15.3 (19D76)

### Additional context

kittaakos, 2020-04-07:
There is an issue with `Arduino Zero` too:
```
arduino-cli.exe core search "Arduino SAMD Boards (32" --format json
```
Before install:
```json
      {
        "name": "Arduino Zero"
      },
```

After the corresponding core installation:
```json
      {
        "name": "Arduino Zero (Native USB Port)",
        "fqbn": "arduino:samd:arduino_zero_native"
      },

      {
        "name": "Arduino Zero (Programming Port)",
        "fqbn": "arduino:samd:arduino_zero_edbg"
      }
```

ubidefeo, 2020-04-07:
@kittaakos 
the Arduino Zero has 2 USB ports: one is a USB<>UART which allows uploading firmware using the chip's bootloader, the other one connects to Atmel's EDBG chip (a debugger bridge).
I think the old version of the core did not allow programming through EDBG, that's why you see two.

The Java IDE also presents both :)

kittaakos, 2020-04-07:
Thanks for the update, @ubidefeo. I think the name of the board still should not change, but both should be declared without the FQBNs: `Arduino Zero (Native USB Port)` and `Arduino Zero (Programming Port)`, and when I install it, the FQBNs will be set. This does not happen now.

> The Java IDE also presents both :)

This issue is not about whether it is there twice or not, but the name change.

per1234, 2020-04-14:
The differences in names are often intentional. It's going to be challenging to achieve compliance with this requirement in the 3rd party platform realm. It's also likely that compliance will break over time, since you must keep these two sets of strings in sync as boards are added or names are changed. For this reason, mismatches should be expected and handled gracefully by Arduino Pro IDE.

If this is something that will be required, I suggest documenting that in the [package_index.json specification](https://arduino.github.io/arduino-cli/package_index_json-specification/) so at least we will have some hope that new platforms will start in compliance.

per1234, 2022-02-22:
Closing as resolved because this inevitable mismatch between the value of the `packages[*].platforms[*].boards[*].name` field of the [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/) and the `BOARD_IDE.name` value of the [`boards.txt`](https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt) ` is now handled in a graceful and user friendly manner by Arduino IDE 2.x:

[image]

closed 2022-02-22 by per1234