arduino/arduino-cli #622: automatically download STM32CubeProgrammer

aster94, 2020-03-15:
Hello,

i installed the STM32:stm32 core. I would expect arduino-cli to manage all the dependency
but
```
ino upload -b STM32:stm32:GenF1 -p COM1
STM32_Programmer_CLI.exe not found.
Please install it or add <STM32CubeProgrammer path>\bin' to your PATH environment:
https://www.st.com/en/development-tools/stm32cubeprog.html
Aborting!
Error during Upload: uploading error: exit status 1
```

Really it is a little problem because the error message is perfect and explain exactly what i am supposed to do, but still i would suggest to the team to automaticcaly download it
https://www.st.com/en/development-tools/stm32cubeprog.html

per1234, 2020-03-15:
This isn't about a limitation or malfunction of arduino-cli. The authors of STM32 simply didn't provide the tool in the boards package:
https://github.com/stm32duino/wiki/wiki/Upload-methods#requirement
>To use those upload methods, STM32CubeProgrammer have to be installed manually as it is not provided through the tools packages.

So this would be something to take up with the stm32duino folks. Arduino already provides the mechanisms for automatically installing tools dependencies with a platform. It's up to the platform author whether they want to take advantage of it or not. There's nothing more that can be done in arduino-cli to help with this.

closed 2020-03-15 by per1234

aster94, 2020-03-15:
Clear!