arduino/arduino-cli #623: [question] Retrieve 3rd party board's name/FQBN with `board search`

kittaakos, 2020-03-16:
Is it possible to retrieve the name, maybe the FQBN of the attached 3rd party board?

This is what I have done:
 - Build a CLI from the `HEAD` of the `master` (79bed420) and wipe any previous state:
  ```
akos.kitta@Akoss-MacBook-Pro arduino-cli % git rev-parse --short HEAD 
79bed420
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli version
arduino-cli Version: 0.0.0-git Commit: 
akos.kitta@Akoss-MacBook-Pro arduino-cli % rm -rf ~/Library/Arduino15 && rm -rf ~/Documents/Arduino 
  ```
 - Initialize a CLI config and add an additional 3rd party boards package URL:
  ```
Config file written to: arduino-cli.yaml
akos.kitta@Akoss-MacBook-Pro arduino-cli % cat ./arduino-cli.yaml 
board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: /Users/akos.kitta/Library/Arduino15
  downloads: /Users/akos.kitta/Library/Arduino15/staging
  user: /Users/akos.kitta/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
akos.kitta@Akoss-MacBook-Pro arduino-cli % vi ./arduino-cli.yaml 
akos.kitta@Akoss-MacBook-Pro arduino-cli % cat ./arduino-cli.yaml
board_manager:
  additional_urls: [
    "https://arduino.esp8266.com/stable/package_esp8266com_index.json"
  ]
daemon:
  port: "50051"
directories:
  data: /Users/akos.kitta/Library/Arduino15
  downloads: /Users/akos.kitta/Library/Arduino15/staging
  user: /Users/akos.kitta/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
  ```
 - Update the index  and verify the 3rd party boards package is available:
  ```
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli core update-index 
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded                                                                                                                       
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded                                                                                                                   
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_esp8266com_index.json downloaded                                                                                                        
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli core search "esp"
ID              Version Name              
arduino:avr     1.8.2   Arduino AVR Boards
esp8266:esp8266 2.6.3   esp8266           
  ```
 - Install the desired, `esp8266:esp8266` 3rd party core:
  ```
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli core install esp8266:esp8266
Downloading packages...
esp8266:xtensa-lx106-elf-gcc@2.5.0-4-b40a506 downloaded                                                                                                         
esp8266:mkspiffs@2.5.0-4-b40a506 downloaded                                                                                                                     
esp8266:mklittlefs@2.5.0-4-69bd9e6 downloaded                                                                                                                   
esp8266:python3@3.7.2-post1 downloaded                                                                                                                          
esp8266:esp8266@2.6.3 downloaded                                                                                                                                
Installing esp8266:xtensa-lx106-elf-gcc@2.5.0-4-b40a506...
esp8266:xtensa-lx106-elf-gcc@2.5.0-4-b40a506 installed
Installing esp8266:mkspiffs@2.5.0-4-b40a506...
esp8266:mkspiffs@2.5.0-4-b40a506 installed
Installing esp8266:mklittlefs@2.5.0-4-69bd9e6...
esp8266:mklittlefs@2.5.0-4-69bd9e6 installed
Installing esp8266:python3@3.7.2-post1...
esp8266:python3@3.7.2-post1 installed
Installing esp8266:esp8266@2.6.3...
esp8266:esp8266@2.6.3 installed
  ```
 - Attach the board and list them:
  ```
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli board list                  
Port                            Type              Board Name FQBN Core
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown             
/dev/cu.MALS                    Serial Port       Unknown             
/dev/cu.SLAB_USBtoUART          Serial Port (USB) Unknown             
/dev/cu.SOC                     Serial Port       Unknown             

akos.kitta@Akoss-MacBook-Pro arduino-cli %
  ```

The board name and the FQBN are `Unknown`. Is it a bug or that's how it is and not all board names/FQBNs can be retrieved although the corresponding boards package is installed.

Note: The board works, I can verify and upload. The `serial-discovery` can also see it:

```
akos.kitta@Akoss-MacBook-Pro arduino-cli % ~/Library/Arduino15/packages/builtin/tools/serial-discovery/1.0.0/serial-discovery 
LIST
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/cu.SOC",
      "label": "/dev/cu.SOC",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.MALS",
      "label": "/dev/cu.MALS",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.Bluetooth-Incoming-Port",
      "label": "/dev/cu.Bluetooth-Incoming-Port",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.SLAB_USBtoUART",
      "label": "/dev/cu.SLAB_USBtoUART",
      "prefs": {
        "productId": "0xEA60",
        "serialNumber": "0001",
        "vendorId": "0x10C4"
      },
      "identificationPrefs": {
        "pid": "0xEA60",
        "vid": "0x10C4"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

OS: macOS `10.15.3 (19D76)`
CLI: Built from the source: 79bed420

per1234, 2020-03-16:
The FQBN can only be determined if the VID/PID of the port matches a the VID/PID definition of one of the boards in boards.txt. AFAIK, the ESP8266 boards all use general purpose USB to TTL serial adapter chips (e.g., FTDI FT232, WCH CH340, Silicon Labs CP2102/CP2104, Prolific PL2303) with the manufacturer's stock VID/PID. For this reason, no VID/PID pairs are defined in the ESP8266 platform's boards.txt.

cmaglie, 2020-03-16:
Yes, we basically see the VID/PID of the general-purpose USB-2-UART converter chip, not a specific VID/PID of the board.
In other word we know that the board uses the USB-UART chip but we don't know which board is.

kittaakos, 2020-03-16:
I see. Is there a chance to at least guess which boards package the attached board belongs to?
I want to give some hint to the user, something like:
```
Unknown board [from ${guesseedPackageName}] connected to /dev/cu.SLAB_USBtoUART
```

If no, please close the issue and thank you for the explanation 😊

matthijskooijman, 2020-09-11:
I think this is not possible if the board uses a generic vid/pid. One could guess that it is an esp board, but there might very well be other boards/cores that *also* use the same generic vid/pid.

The only option I could see is that *all* boards that use this vidpid specify it in their boards.txt, then `board list` will list them all. But if it's more than just a handful of boards, the output gets useless quickly (even worse for the Java IDE, which I think just shows one match). `arduino-cli boards list` could maybe be smart about this and if > 5 boards or so, summarize them to "25 boards from package esp8266", but I think that would get too complex. I guess this is just a limitation of not using a unique USB id for a board.

I'd suggest closing this issue indeed.

ubidefeo, 2020-09-11:
agreed, @matthijskooijman 
I'll close it :)

closed 2020-09-11 by ubidefeo