arduino/arduino-cli #629: [feature-request]: Support fuzzy `core search`

kittaakos, 2020-03-21:
Inspired by https://github.com/arduino/arduino-pro-ide/issues/232.

In the Arduino Pro IDE, I want to return the `Arduino Duemilanove or Diecimila` board when the user searches `ar du`. Currently, it is not possible.

Workaround: run a wildcard search, get back the board names (`--format json`, or gRPC), then run a custom fuzzy search.

noob-master147, 2020-04-03:
hey @kittaakos, this seems interesting.
I would like to look into this 😃

any tips from where I shall start?

noob-master147, 2020-04-05:
Can I get a small help?

    resp, err := core.PlatformSearch(inst.GetId(), arguments, allVersions)

**which file is the func PlatformSearch() defined in?**
@kittaakos

umbynos, 2023-12-07:
Seems to have been added: 
```
$ arduino-cli core search "ar du" --format json | grep -i duemilanove
        "name": "Arduino Duemilanove or Diecimila",
```