arduino/arduino-cli #630: "compile --build-path" does not work for relative paths

ktdreyer, 2020-03-23:
I cannot specify a relative path for `compile --build-path`.

In other words, this does not work:
```
arduino-cli compile -b adafruit:avr:protrinket3 . -v --build-path $PWD/_build
[... snip ...]
Linking everything together...
Error during build: Rel: can't make /home/ktdreyer/dev/arduino/_build/core/core.a relative to _build
```

It took me a while to understand this error message.

It does work when I specify an absolute path:
```
arduino-cli compile -b adafruit:avr:protrinket3 . -v --build-path $PWD/_build
```

It would be great to make the following improvements:
1) (Easy) Update the help text in `compile --help` to indicate that `--build-path` must be an absolute path.
2) (Harder) Support relative paths for `--build-path`.

vinay-lanka, 2020-03-23:
Hey, 
Did you mean &nbsp; `arduino-cli compile -b adafruit:avr:protrinket3 . -v --build-path./_build ` does not work?

ubidefeo, 2020-03-23:
hi @ktdreyer 
I just tested and it correctly created a `_build` folder, compiled the local sketch into it and returned success.
What's your configuration?

ktdreyer, 2020-03-23:
My bad, I screwed up my initial report here. I meant to say that the command does not work when I omit "`$PWD`".

I tried "`./_build`" and "`_build`", and neither work.

I tested this on my normal Fedora 31 laptop as well as a completely new `fedora:31` container. It always fails at this step:

```
Error during build: Rel: can't make /tmp/MyFirstSketch/_build/core/core.a relative to ./_build
```

ubidefeo, 2020-03-23:
ok, better now
it fail for me too, obviously :)

Honestly I'm used to extracting absolute paths in my shell scripts as well as in other languages, so don't really feel strong about this, but I can see it might be useful.

what do @rsora and @masci think about it?

yzykov, 2020-03-29:
Hi,

I have a concern about the issue itself & [proposed fix](https://github.com/arduino/arduino-cli/pull/632). According to other functions in the CLI library the path have to be absolute, e.g.:
    1) [Upload](https://github.com/arduino/arduino-cli/blob/e8bb3278bd11d5cd351ef102526dd36980d6f20a/commands/upload/upload.go#L49) and its [documentation](https://arduino.github.io/arduino-cli/commands/arduino-cli_upload/) it should have similar issue
    2) [Debug](https://github.com/arduino/arduino-cli/blob/e8bb3278bd11d5cd351ef102526dd36980d6f20a/commands/debug/debug.go#L123)
and [here](https://github.com/arduino/arduino-cli/blob/e8bb3278bd11d5cd351ef102526dd36980d6f20a/commands/debug/debug.go#L199)
with [similar description](https://arduino.github.io/arduino-cli/commands/arduino-cli_debug/)

and even in [Compile](https://github.com/arduino/arduino-cli/blob/e8bb3278bd11d5cd351ef102526dd36980d6f20a/commands/compile/compile.go#L154) for Cache Path. 

Why do you want to change compile --build-path functionality, excluding the others?
Therefore, I am suggesting to change documentation rather than fixing compile --build-path, which will cause ambiguity.

vinay-lanka, 2020-04-04:
Makes sense, we could actually make a small modification to the way paths are handled by allowing support for both absolute and relative paths.
It seems more convenient to have the option to use either.
What do you guys think?

ktdreyer, 2020-08-17:
I think users will expect both absolute and relative paths to work. Like when I pass an absolute or relative path to `ls`, it does what I expect.

bxparks, 2020-09-03:
@ubidefeo: Just wanted to say thanks, the `compile --upload` flag worked well to work around the relative path problem with the `upload` command.

ubidefeo, 2020-09-10:
@bxparks 
good to hear
by the way in the nightly build you should find the `--input-file` working again.
Of course you'll need to pass in the `--fqbn` but it works great

josejsarmento, 2021-08-31:
This is still happening on Mac OS 11.5.2 and arduino-cli 0.18.3

## Bug Report

### Current behavior

Compiling a sketch with the `--build-path` flag is not working for me. Whenever I run the following command:

`arduino-cli compile --build-path ../ArduinoOutput -b esp8266:esp8266:generic sketch`

I get:

```
Couldn't deeply cache core build: Rel: can't make ../ArduinoOutput relative to /var/folders/gg/v110hgf95ls12d5vvmv2mc0c0000gn/T/arduino-core-cache
Running normal build of the core...
```

### Expected behavior

I expect the program to save the compiled files at the chosen folder and to re-use them when I compile again afterwards. This would be similar to the 'output' option in the Arduino extension in Visual Studio Code.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-21T17:11:27Z`
- OS and platform: Mac OS 11.5.2

closed 2021-08-31 by silvanocerza