arduino/arduino-cli #631: Feature request: exclude files from Arduino CLI

janjongboom, 2020-03-23:
I was contemplating whether this would be a request for the library format or something in Arduino CLI. We have a unified codebase for our embedded libraries, and deliver a configuration file per target that includes / excludes certain folders. E.g. we can deliver a Makefile, a Keil uVision project, or an `.mbedignore` file as part of the delivery mechanism. This works really well. However, for Arduino there does not seem to be a similar way to exclude files from the build process. I know others struggle with this too (e.g. @neil-tan when developing uTensor). 

Adding this in the library format and in Arduino CLI would make it a lot easier to distribute upstream versions of libraries that we publish, rather than having to create separate builds just for Arduino.

alranel, 2020-05-14:
Hi @janjongboom!
If I understand correctly you're thinking about something like a couple of `--include` and `--exclude` options to tell the compiler which .c, .cpp and .ino files to consider. Did I get it right?

If so, maybe is this something that can be achieved with a clever Makefile that copies the relevant files/folders for the selected target to a temporary build directory and then launches arduino-cli?

janjongboom, 2020-05-18:
@alranel Correct, and yes, that's one way of doing it, e.g. this is how we're doing it right now:

https://github.com/edgeimpulse/inferencing-sdk-cpp/blob/master/create-arduino-library.sh

But that always requires a second step, e.g. we need to do this every time someone wants to include our library as an Arduino lib.

alranel, 2020-05-19:
What about setting up a CI action on your repo, that performs that cleanup and pushes the Arduino library to its own repo? That way users will have it ready like a normal Arduino library, always up-to-date with your upstream repo. Also you could include a library.properties file in there so that it can be indexed in the Arduino Library Manager.
Would this work for you?

per1234, 2020-05-27:
Hi @janjongboom. I'd be glad to assist with implementing alranel's proposal of using CI to generate the Arduino library from your unified code.

My idea is that on every push to the edgeimpulse/inferencing-sdk-cpp repository's `master` branch that modifies relevant files, the CI system should:
1. Run an expanded version of create-arduino-library.sh that takes all the steps to product a valid Arduino library.
1. Do a compilation test using Arduino CLI to make sure the code will compile with the Arduino development software.
1. If the test passes, push the updated library to a dedicated repository.

janjongboom, 2020-05-28:
@per1234 Thanks, we already have this in our internal CI (and automatically test building a bunch of impulses for the Nano 33 BLE Sense), but it's still annoying that the Arduino CLI cannot do this yet.

per1234, 2020-06-05:
I'm glad to hear you have been able to automate this @janjongboom.

We will investigate whether the exclude functionality can be provided within a broader context as part of the ongoing development of Arduino CLI.

alranel, 2021-08-02:
In case anyone is interested in this, I'm sure the Arduino team would be open to review any contribution that implements it :)