arduino/arduino-cli #636: Error during install of arduino:avr.

mbrooks-UoA, 2020-03-28:
Hello all,

While trying to get a fresh install set up to run with my Arduino Mega 2560 I encountered and error when I tried to install the arduino:avr core using the command "C:\Users\[UserName]\[rest of file path]>arduino-cli core install arduino:avr" in the cmd.

I got this error: "Error during install: moving extracted archive to destination dir: rename C:\Users\[UserName]\AppData\Local\Arduino15\tmp\package-586261211\avr C:\Users\[UserName]\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2: Access is denied."
 
Also, if it is helpful the package I downloaded and tried to install had this in it "package-586261211\avr\firmwares\wifisheild\wifi_dnld\src\docs\SPB104 product brief.pdf" 
I thought it was kind of weird that I downloaded a PDF and I thought it might be good to include. 

I am using windows 10 and I'm fairly new to using cmd, so I came here to ask for some advice. If anyone could provide some assistance on how to get the core to install I would greatly appreciate it.

per1234, 2020-03-28:
>Access is denied

This sort of error could be caused by your antivirus software. Try **TEMPORARILY** disabling your antivirus for a single compilation to see if the problem goes away, then turn the antivirus back on. If the problem doesn't occur with the antivirus off you will need to adjust the settings of your antivirus to whitelist the appropriate file, folder, or process so it doesn't interfere with compilation.

>I thought it was kind of weird that I downloaded a PDF

This is normal and expected. That file is present in the Arduino AVR Boards platform:
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/wifishield/wifi_dnld/src/Doc

rsora, 2020-04-01:
Hi @mbrooks-UoA !
Do the suggestions made by @per1234 fix your problem?

Thanks for reaching out!

mbrooks-UoA, 2020-04-02:
I'm sorry for the late reply, per1234's suggestion worked and I was able to install the core needed. Thank you for the help per1234, you are a life saver.

closed 2020-04-02 by mbrooks-UoA