arduino/arduino-cli #637: Impossible to set multiple options for a board

ilgiznurgaliev, 2020-03-28:
## Bug Report

### Current behavior
```
$ arduino-cli board details "LGT8fx Boards:avr:328"
Board name: LGT8F328                                           

Option:     Clock Source                                          clock_source                
            Internal                                            ✔ clock_source=internal       
            External (assumes 32MHz crystal)                      clock_source=external       

Option:     Clock                                                 clock                       
            32 MHz                                              ✔ clock=32MHz                 
            16 MHz                                                clock=16MHz                 
            8 MHz                                                 clock=8MHz                  
            4 MHz                                                 clock=4MHz                  
            2 MHz                                                 clock=2MHz                  
            1 MHz                                                 clock=1MHz                  

Option:     Variant                                               variant                     
            328P-LQFP48 MiniEVB                                 ✔ variant=modelP48            
            328P-LQFP32 (e.g. MiniEVB nano-style)                 variant=modelP              
            328P-LQFP32 wemos-TTGO-XI                             variant=modelP-wemos-TTGO-XI
            328D (e.g. WAVGAT AVG328D)                            variant=modelD              
            328D-SSOP20 (e.g. green pro mini alternative)         variant=modelD_SSOP20

$ arduino-cli compile -b "LGT8fx Boards:avr:328:clock=16MHz:variant=modelP" test
Error during build: incorrect FQBN: invalid fqbn: LGT8fx Boards:avr:328:clock=16MHz:variant=modelP

$ arduino-cli board attach "LGT8fx Boards:avr:328:clock=16MHz:variant=modelP" test
Attach board error: invalid Device URL format: parse "serial://LGT8fx Boards:avr:328:clock=16MHz:variant=modelP": invalid port ":variant=modelP" after host
```

**Getting the error above when trying to set two options (clock and variant in this case) for the board.**

### Expected behavior
Ability to set multiple board options.

### Environment

- CLI version (output of `arduino-cli version`): **arduino-cli Version: 0.9.0-arch Commit: e4d02b45a8e5777e701a786e18f55c1a63753ce6**
- OS and platform: **ArchLinux with kernel 5.5.13-arch1-1**

rsora, 2020-04-02:
Hi @ilgiznurgaliev, thanks for reaching out!

Could you please try to separate with **comma** the additional options?
Try something like  `arduino-cli compile -b  "LGT8fx Boards:avr:328:clock=16MHz,variant=modelP" test`

We understand that this is a feature not perfectly documented, would you be so nice to open a PR in our [FAQ documentation section](https://github.com/arduino/arduino-cli/blob/master/docs/FAQ.md) to help also other users that will encouter your same issue?

Happy coding!

ilgiznurgaliev, 2020-04-02:
Hi Roberto.

Works as expected with commas. 
PR #638 has been created to update FAQ.

Thank you very much.

closed 2020-04-02 by ilgiznurgaliev