arduino/arduino-cli #641: Upload build path is different than output build path

ardnew, 2020-04-02:
## Bug Report

### Current behavior

```sh
arduino-cli --config-file /home/andrew/.config/arduino-cli/config.yaml compile --verbose --log-level debug --fqbn adafruit:nrf52:itsybitsy52840 --upload --port /dev/ttyACM1  --build-path /home/andrew/.tmp/arduino-build/bled --build-cache-path /home/andrew/.tmp/arduino-build.cache/bled  /home/andrew/Development/arduino/sketchbook/bled
```

The Adafruit nRF52840 core platform.txt contains a recipe to generate a bootable DFU .zip package specifying the output path as `{build.path}/{build.project_name}.zip`:
```sh
## Create dfu package zip file
recipe.objcopy.zip.pattern="{tools.nrfutil.cmd}" dfu genpkg --dev-type 0x0052 --sd-req {build.sd_fwid} --application "{build.path}/{build.project_name}.hex" "{build.path}/{build.project_name}.zip"
```

The upload utility recipe uses the exact same path `{build.path}/{build.project_name}.zip`:
```sh
tools.nrfutil.upload.pattern="{cmd}" {upload.verbose} dfu serial -pkg "{build.path}/{build.project_name}.zip" -p {serial.port} -b 115200 --singlebank
```

Yet when I execute the command pasted above (with a build path given at the command line), the outputs indicate two different paths are used:
```sh
...
Running recipe: recipe.objcopy.zip.pattern
adafruit-nrfutil dfu genpkg --dev-type 0x0052 --sd-req 0x00B6 --application /home/andrew/.tmp/arduino-build/bled/bled.ino.hex /home/andrew/.tmp/arduino-build/bled/bled.ino.zip
Zip created at /home/andrew/.tmp/arduino-build/bled/bled.ino.zip
...
Sketch uses 35652 bytes (4%) of program storage space. Maximum is 815104 bytes.
Global variables use 6612 bytes (2%) of dynamic memory, leaving 230956 bytes for local variables. Maximum is 237568 bytes.
No new serial port detected.
Usage: adafruit-nrfutil dfu serial [OPTIONS]

Error: Invalid value for "-pkg" / "--package": Path "/home/andrew/Development/arduino/sketchbook/bled/bled.adafruit.nrf52.itsybitsy52840.zip" does not exist.
Error during Upload: uploading error: exit status 2
```

If I modify either path pattern I referenced above in platform.txt, I can verify the output of both commands uses their respective modified paths, so I'm confident these are the recipes being used.

### Expected behavior

I would expect the path pattern `{build.path}/{build.project_name}.zip` to resolve to the same path in all recipes, and in particular that path should reflect what's given by command line option `--build-path`.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.0.0-git Commit: 8759bf46
- OS and platform: Ubuntu 18.04.4 LTS

### Additional context

cmaglie, 2020-09-14:
@ardnew 
we merged a lof of fixes about build paths, may you try the latest version 0.13.0 and see if this is fixed?

ardnew, 2020-09-14:
@cmaglie sure thing, I'll try again this evening. are you expecting the other bug (#746) might be addressed as well, or no?

cmaglie, 2020-09-15:
It should... give it a try and let us know :-)

closed 2020-12-15 by github-actions[bot]