arduino/arduino-cli #643: Protoc : Can't compile protocol buffers

vinay-lanka, 2020-04-06:
## Bug Report

### Current behavior

#### Command
Running `task protoc` or
`protoc --proto_path=rpc --go_out=plugins=grpc,paths=source_relative:rpc ./rpc/commands/*.proto` seems to throw an error 
```
$ task protoc
protoc --proto_path=rpc --go_out=plugins=grpc,paths=source_relative:rpc ./rpc/commands/*.proto
--go_out: protoc-gen-go: plugins are not supported; use 'protoc --go-grpc_out=...' to generate gRPC
task: Failed to run task "protoc": exit status 1
```

### Expected behavior
Expected generation of protocol buffer definitions
Am I doing something wrong?

According to [this](https://github.com/protocolbuffers/protobuf-go/releases/tag/v1.20.0#v1.20-grpc-support) - 

The v1.20 protoc-gen-go does not support generating gRPC service definitions. In the future, gRPC service generation will be supported by a new protoc-gen-go-grpc plugin provided by the Go gRPC project.

### Environment

- CLI version (output of `arduino-cli version`):
Commit: 4096f524
- OS and platform:
Currently on Ubuntu 18.04 LTS

rsora, 2020-04-07:
Hi! I cannot reproduce your issue.
Can you post output of `protoc --version`?

Thanks!

vinay-lanka, 2020-04-07:
```
$ protoc --version
libprotoc 3.0.0
```

rsora, 2020-04-07:
I think you should use the latest version, check [here](https://askubuntu.com/a/1072684) how to install it!

vinay-lanka, 2020-04-07:
Oh got it. It works now thanks!

closed 2020-04-07 by vinay-lanka