arduino/arduino-cli #644: `serial-discovery LIST` lists an attached 3rd party board twice

kittaakos, 2020-04-07:
## Bug Report

My `ESP8266` 3rd party board is listed twice via the `serial-discovery`. One can see the same `productId`, `serialNumber`, and `vendorId` triplet is listed twice on separate USB ports.

```json
{
  "address": "/dev/cu.SLAB_USBtoUART",
  "label": "/dev/cu.SLAB_USBtoUART",
  "prefs": {
    "productId": "0xEA60",
    "serialNumber": "0001",
    "vendorId": "0x10C4"
  },
  "identificationPrefs": {
    "pid": "0xEA60",
    "vid": "0x10C4"
  },
  "protocol": "serial",
  "protocolLabel": "Serial Port (USB)"
},
{
  "address": "/dev/cu.usbserial-0001",
  "label": "/dev/cu.usbserial-0001",
  "prefs": {
    "productId": "0xEA60",
    "serialNumber": "0001",
    "vendorId": "0x10C4"
  },
  "identificationPrefs": {
    "pid": "0xEA60",
    "vid": "0x10C4"
  },
  "protocol": "serial",
  "protocolLabel": "Serial Port (USB)"
}
```

### Current behavior

Steps to reproduce:

 1. Wipe the previous state:
```
rm -rf ~/Library/Arduino15 && rm -rf ~/Documents/Arduino
```

 2. Check version:
```
./arduino-cli version
arduino-cli Version: 0.9.0 Commit: e4d02b4
```

 3.  Update index with the 3rd party core:
```
./arduino-cli core update-index --additional-urls https://dl.espressif.com/dl/package_esp32_index.json
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded                                                                                                                       
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded                                                                                                                   
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_esp32_index.json downloaded                                                                                                             
```

 4. Verify 3rd party core is visisble:
```
./arduino-cli core search esp --additional-urls https://dl.espressif.com/dl/package_esp32_index.json
ID          Version Name              
arduino:avr 1.8.2   Arduino AVR Boards
esp32:esp32 1.0.4   esp32             
```

 5.  Install the 3rd party core:
```
./arduino-cli core install esp32:esp32 --additional-urls https://dl.espressif.com/dl/package_esp32_index.json
Downloading packages...
esp32:xtensa-esp32-elf-gcc@1.22.0-80-g6c4433a-5.2.0 downloaded                                                                                                  
esp32:esptool_py@2.6.1 downloaded                                                                                                                               
esp32:mkspiffs@0.2.3 downloaded                                                                                                                                 
esp32:esp32@1.0.4 downloaded                                                                                                                                    
Installing esp32:xtensa-esp32-elf-gcc@1.22.0-80-g6c4433a-5.2.0...
esp32:xtensa-esp32-elf-gcc@1.22.0-80-g6c4433a-5.2.0 installed
Installing esp32:esptool_py@2.6.1...
esp32:esptool_py@2.6.1 installed
Installing esp32:mkspiffs@0.2.3...
esp32:mkspiffs@0.2.3 installed
Installing esp32:esp32@1.0.4...
esp32:esp32@1.0.4 installed
```

 6. Verify core installation:
```
./arduino-cli core list --additional-urls https://dl.espressif.com/dl/package_esp32_index.json
ID          Installed Latest Name 
esp32:esp32 1.0.4     1.0.4  esp32
```

 7. Attach the ESP8266 board and list the attached boards with the CLI:
```
./arduino-cli board list --additional-urls https://dl.espressif.com/dl/package_esp32_index.json
Port                            Type              Board Name FQBN Core
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown             
/dev/cu.MALS                    Serial Port       Unknown             
/dev/cu.SLAB_USBtoUART          Serial Port (USB) Unknown          <-- This USB port referrs to the same board as the one on  '/dev/cu.usbserial-0001'.
/dev/cu.SOC                     Serial Port       Unknown             
/dev/cu.usbserial-0001          Serial Port (USB) Unknown          <-- Same as the one from the '/dev/cu.SLAB_USBtoUART' port.
```

8. Call the `serial-discovery` directly:
```
~/Library/Arduino15/packages/builtin/tools/serial-discovery/1.0.0/serial-discovery 
LIST
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/cu.MALS",
      "label": "/dev/cu.MALS",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.SOC",
      "label": "/dev/cu.SOC",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.Bluetooth-Incoming-Port",
      "label": "/dev/cu.Bluetooth-Incoming-Port",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.SLAB_USBtoUART",
      "label": "/dev/cu.SLAB_USBtoUART",
      "prefs": {
        "productId": "0xEA60",
        "serialNumber": "0001",
        "vendorId": "0x10C4"
      },
      "identificationPrefs": {
        "pid": "0xEA60",
        "vid": "0x10C4"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "/dev/cu.usbserial-0001",
      "label": "/dev/cu.usbserial-0001",
      "prefs": {
        "productId": "0xEA60",
        "serialNumber": "0001",
        "vendorId": "0x10C4"
      },
      "identificationPrefs": {
        "pid": "0xEA60",
        "vid": "0x10C4"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

### Expected behavior

The board is listed once. I am pretty sure that was the case previously, maybe something has changed on OS level.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.9.0 Commit: e4d02b4`
- OS and platform: macOS `10.15.4 (19E266)`

### Additional context

kittaakos, 2020-04-07:
Once I was told to install the USB driver for macOS from this site for my CP2104 board because it was not recognized on OS level: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

Today, I have reinstalled the latest `v5.2.4` version of the same driver. It did not help. Then I decided to wipe it from my macOS. After the driver uninstallation, I do not see duplicate drivers.

```
akos.kitta@Akoss-MacBook-Pro cli-test % ./arduino-cli board list
Port                            Type              Board Name FQBN Core
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown             
/dev/cu.MALS                    Serial Port       Unknown             
/dev/cu.SOC                     Serial Port       Unknown             
/dev/cu.usbserial-0001          Serial Port (USB) Unknown             

akos.kitta@Akoss-MacBook-Pro cli-test % ~/Library/Arduino15/packages/builtin/tools/serial-discovery/1.0.0/serial-discovery 
LIST
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/cu.SOC",
      "label": "/dev/cu.SOC",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.MALS",
      "label": "/dev/cu.MALS",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.Bluetooth-Incoming-Port",
      "label": "/dev/cu.Bluetooth-Incoming-Port",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.usbserial-0001",
      "label": "/dev/cu.usbserial-0001",
      "prefs": {
        "productId": "0xEA60",
        "serialNumber": "0001",
        "vendorId": "0x10C4"
      },
      "identificationPrefs": {
        "pid": "0xEA60",
        "vid": "0x10C4"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

kittaakos, 2020-04-07:
I am experiencing the same oddity when I attach my Arduino Zero board with the [Black Magic Probe V2.1](https://1bitsquared.com/products/black-magic-probe).

I can see three attached boards although it should be two only:
```
/arduino-cli board list
Port                            Type              Board Name                      FQBN                           Core        
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                                                    
/dev/cu.MALS                    Serial Port       Unknown                                                                    
/dev/cu.SOC                     Serial Port       Unknown                                                                    
/dev/cu.usbmodem14602           Serial Port (USB) Arduino Zero (Programming Port) arduino:samd:arduino_zero_edbg arduino:samd
/dev/cu.usbmodemD6E4A0B61       Serial Port (USB) Unknown                                                                    
/dev/cu.usbmodemD6E4A0B63       Serial Port (USB) Unknown
```

And when I use the `serial-discovery` directly, I see the Black Magic Probe twice:
```
~/Library/Arduino15/packages/builtin/tools/serial-discovery/1.0.0/serial-discovery 
LIST
{
  "eventType": "list",
  "ports": [
    {
      "address": "/dev/cu.SOC",
      "label": "/dev/cu.SOC",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.MALS",
      "label": "/dev/cu.MALS",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.Bluetooth-Incoming-Port",
      "label": "/dev/cu.Bluetooth-Incoming-Port",
      "prefs": {},
      "identificationPrefs": {},
      "protocol": "serial",
      "protocolLabel": "Serial Port"
    },
    {
      "address": "/dev/cu.usbmodem14602",
      "label": "/dev/cu.usbmodem14602",
      "prefs": {
        "productId": "0x2157",
        "serialNumber": "3ZXXH6CNQ4ME0E3Q69HT",
        "vendorId": "0x03EB"
      },
      "identificationPrefs": {
        "pid": "0x2157",
        "vid": "0x03EB"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "/dev/cu.usbmodemD6E4A0B63",
      "label": "/dev/cu.usbmodemD6E4A0B63",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "D6E4A0B6",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "/dev/cu.usbmodemD6E4A0B61",
      "label": "/dev/cu.usbmodemD6E4A0B61",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "D6E4A0B6",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

Same as from above:
```json
    {
      "address": "/dev/cu.usbmodemD6E4A0B63",
      "label": "/dev/cu.usbmodemD6E4A0B63",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "D6E4A0B6",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "/dev/cu.usbmodemD6E4A0B61",
      "label": "/dev/cu.usbmodemD6E4A0B61",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "D6E4A0B6",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
```

cmaglie, 2020-04-07:
are those /dev files real?

```
/dev/cu.usbmodemD6E4A0B63
/dev/cu.usbmodemD6E4A0B61
```

if they are present in the `/dev` folder then the serial_discovery is reporting them correctly... It may be that you have two different drivers installed as fot the CP2104?
Maybe the "system info" (menu Apple -> About this mac -> "more info" -> USB) can give some light.

kittaakos, 2020-04-07:
> are those /dev files real?
> 
> ```
> /dev/cu.usbmodemD6E4A0B63
> /dev/cu.usbmodemD6E4A0B61
> ```

I do not know what you mean by "real" but I can see the followings with `ls -al` under `/dev/`:
```
...
crw-rw-rw-   1 root        wheel           18,  10 Apr  7 11:31 tty.usbmodem14602
crw-rw-rw-   1 root        wheel           18,   8 Apr  7 11:09 tty.usbmodemD6E4A0B61
crw-rw-rw-   1 root        wheel           18,   6 Apr  7 11:09 tty.usbmodemD6E4A0B63
...
crw-rw-rw-   1 root        wheel           18,  11 Apr  7 11:31 cu.usbmodem14602
crw-rw-rw-   1 root        wheel           18,   9 Apr  7 11:09 cu.usbmodemD6E4A0B61
crw-rw-rw-   1 root        wheel           18,   7 Apr  7 11:09 cu.usbmodemD6E4A0B63
...
```

This is what I macOS sees:
```
system_profiler SPUSBDataType
2020-04-07 12:40:03.189 system_profiler[16010:89427] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4 
      PCI Revision ID: 0x0002 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x01 

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4 
      PCI Revision ID: 0x0002 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x00 

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCISPT
      PCI Device ID: 0xa12f 
      PCI Revision ID: 0x0031 
      PCI Vendor ID: 0x8086 

        EDBG CMSIS-DAP:

          Product ID: 0x2157
          Vendor ID: 0x03eb  (Atmel Corporation)
          Version: 1.01
          Serial Number: 3ZXXH6CNQ4ME0E3Q69HT
          Speed: Up to 480 Mb/s
          Manufacturer: Atmel Corp.
          Location ID: 0x14600000 / 9
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

        Black Magic Probe:

          Product ID: 0x6018
          Vendor ID: 0x1d50
          Version: 1.00
          Serial Number: D6E4A0B6
          Speed: Up to 12 Mb/s
          Manufacturer: Black Sphere Technologies
          Location ID: 0x14400000 / 7
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

        Apple T1 Controller:

          Product ID: 0x8600
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 1.01
          Manufacturer: Apple Inc.
          Location ID: 0x14200000
```

And here is the shorter version:
```
ioreg -p IOUSB -w0 | sed 's/[^o]*o //; s/@.*$//' | grep -v '^Root.*'
AppleUSBXHCI Root Hub Simulation
Black Magic Probe
EDBG CMSIS-DAP
Apple T1 Controller
```

kittaakos, 2020-04-07:
I can see the Black Magic Probe listed twice on Windows10 too. It is not a macOS issue. See `COM5` and `COM6` ports:
```
C:\Users\kittaakos\AppData\Local\Arduino15\packages\builtin\tools\serial-discovery\1.0.0>serial-discovery.exe
LIST
{
  "eventType": "list",
  "ports": [
    {
      "address": "COM3",
      "label": "COM3",
      "prefs": {
        "productId": "0x0ACA",
        "serialNumber": "",
        "vendorId": "0x8087"
      },
      "identificationPrefs": {
        "pid": "0x0ACA",
        "vid": "0x8087"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "COM4",
      "label": "COM4",
      "prefs": {
        "productId": "0x0ACA",
        "serialNumber": "",
        "vendorId": "0x8087"
      },
      "identificationPrefs": {
        "pid": "0x0ACA",
        "vid": "0x8087"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "COM5",
      "label": "COM5",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "COM6",
      "label": "COM6",
      "prefs": {
        "productId": "0x6018",
        "serialNumber": "",
        "vendorId": "0x1D50"
      },
      "identificationPrefs": {
        "pid": "0x6018",
        "vid": "0x1D50"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    },
    {
      "address": "COM7",
      "label": "COM7",
      "prefs": {
        "productId": "0x2157",
        "serialNumber": "",
        "vendorId": "0x03EB"
      },
      "identificationPrefs": {
        "pid": "0x2157",
        "vid": "0x03EB"
      },
      "protocol": "serial",
      "protocolLabel": "Serial Port (USB)"
    }
  ]
}
```

kittaakos, 2020-04-07:
> Black Magic Probe listed twice on Windows10 too

> It may be that you have two different drivers installed as fot the CP2104?

This could be, although I used the official uninstaller on my mac. For sure, I do not have any kind of extra drivers on Windows, still, I see the board twice.

cmaglie, 2020-04-07:
I looked around and found that Black Magic Probe has really two serial ports: https://github.com/blacksphere/blackmagic/wiki/Getting-Started#connecting-to-your-computer

quote from the wiki:
> On Linux you can check the kernel log to find the device that was allocated:
>
> user@host ~ $ dmesg
> usb 1-1.1: new full-speed USB device number 58 using ehci-pci
> usb 1-1.1: New USB device found, idVendor=1d50, idProduct=6018
> usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> usb 1-1.1: Product: Black Magic Probe 
> usb 1-1.1: Manufacturer: Black Sphere Technologies
 >usb 1-1.1: SerialNumber: DDE6C7C4
> cdc_acm 1-1.1:1.0: This device cannot do calls on its own. It is not a modem.
> cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
> cdc_acm 1-1.1:1.2: This device cannot do calls on its own. It is not a modem.
> cdc_acm 1-1.1:1.2: ttyACM1: USB ACM device
>
> The first interface provides the GDB server, and the second provides a USB to UART adapter.

kittaakos, 2020-04-07:
So you are saying it is OK to see them twice. Fine with me 😊

I think we can close this item as my initial issue went away after wiping the driver for CP2104 macOS, and two ports from the Black Magic Probe is expected.

closed 2020-04-07 by rsora