arduino/arduino-cli #649: [feature-request] CLI completion for the commands

ericonr, 2020-04-10:
It would be pretty cool if we could have tab auto-completion, similar to git's interface (as an example of something that has several subcommands). I would be interested in trying to start it at least for the Fish shell. This would probably only work for the Linux, *BSD and Mac environments, but I think it would make life much easier there.

Things to be supported:

* Tab completion for commands like `lib` and `core`
* Tab completion for the names of the options that can be passed, like `--fqbn`
* Tab completion for the board names and options and for target names when installing cores, for example (possibly hard to implement, would require parsing some stuff as well)

ericonr, 2020-06-25:
@umbynos thanks for the implementation! Anything else you want to track with this issue?

umbynos, 2020-06-26:
Hello @ericonr! Thanks for the feature request! Actually I added only a basic tab completion, it does not yet complete core names, libs, fqbn, because the team is doing a refactoring regarding fqbn [here](https://github.com/arduino/arduino-cli/pull/656). I'll add the functionality in the near future. In the meanwhile please test the functionality and let us know! :smile:

ericonr, 2020-09-19:
Sorry, I haven't had much time to use `arduino-cli`! From my quick testing, completion has been a great aid for usability, so thanks a lot!

I think the main usability concern for me regarding fqbn was in regards to finding out what options were available. Making those available both as completion and (ideally) some sort of help output would make `arduino-cli` an even more awesome tool.

ubidefeo, 2020-09-19:
@ericonr 
you're welcome, @umbynos did a great job on that :)
we'll look at `fqbn` completion in the near future, but for now if you need available options try the `board details` command

`arduino-cli board details -b arduino:avr:nano`

```
Option:                 Processor                                                                                   cpu
                        ATmega328P                                                   ✔                              cpu=atmega328
                        ATmega328P (Old Bootloader)                                                                 cpu=atmega328old
                        ATmega168                                                                                   cpu=atmega168
```

umbynos, 2021-09-29:
https://github.com/arduino/arduino-cli/pull/1431