arduino/arduino-cli #650: Provide tarball with sources

ElXreno, 2020-04-10:
Hi! Can you provide on your each release tarball with sources with all dependencies like a Syncthing? It will be very helpful at least for me.

rsora, 2020-04-14:
Hi @ElXreno !

Our automated release process already provides the sources, you can find them here for example:
https://github.com/arduino/arduino-cli/archive/0.9.0.tar.gz
for the latest release.

Thanks for reaching out and happy coding!

ElXreno, 2020-04-14:
@rsora I mean provide tarball with vendor for building without access to the internet.
About sources of this repo I know :)

rsora, 2020-04-14:
Interesting, can you tell me more about your use case?
Why do you need an offline build?

ElXreno, 2020-04-14:
@rsora In Fedora all builders can build only without access to the internet. I can unbundle all stuff and include all required libraries as a packages, but here is something like cyclic dependency (not infinity). 
Your program requires `google.golang.org/grpc` version 1.27.0 which requires `github.com/envoyproxy/go-control-plane` version 0.9.1 which requires `google.golang.org/grpc` version 1.23.0. We can't provide multiple versions of one dependency in repository and as workaround we can use tarball with all sources of all dependencies. But is only workaround and need to fix this with any other solution. Can you help me with it?

cmaglie, 2022-10-13:
Hi @ElXreno, sorry for the long delay in replying here.

The go-lang package system allows a project to provide the `go.mod` file with all the dependencies versions required to build the project (and the `go.sum` with the checksums to verify them that is automatically generated): this is the only requirement for a generic go project.

Your requirest to bundle a copy of **all** the source code for offline builds sounds really weird so, sorry, but we are not going to implement it in our CI infrastructure that is already very complex.

I don't know how fedora works, you may create such tarball by yourself using the [`go mod vendor` command](https://golangbyexample.com/vendor-dependency-go/), but I'll state it again, this is not the standard way in go.

closed 2022-10-13 by cmaglie