arduino/arduino-cli #651: add option to disable color escapes on stdout

nerdralph, 2020-04-12:
## Bug Report

### Current behavior

arduino-cli board details

color escape characters show up on stdout:
Option:     MCU          mcu
            ←[32mATtiny13←[0m   ←[32m?←[0m ←[32mmcu=t13←[0m

### Expected behavior

no escape codes

### Environment

- CLI version (output of `arduino-cli version`): 0.9.0 Commit: e4d02b4
- OS and platform: Win7E/mingw32

### Additional context

colored output on stdout appears to be hard-coded.
https://github.com/arduino/arduino-cli/blob/2791756cec59cd38c71d6facbca5a29e71c37e40/cli/cli.go#L180

HowJMay, 2020-06-13:
Is there a way to reproduce the colored message without Arduino board?

per1234, 2020-06-13:
@HowJMay this should do it (the last command produces the color escape output, the first two are just needed to install the `arduino:avr` platform so that the last command will work):
```
arduino-cli core update-index
arduino-cli core install arduino:avr
arduino-cli board details arduino:avr:nano
```

HowJMay, 2020-06-14:
May I try to take this issue. I guess I find the solution for this issue (at least the command listed above). What is the command that can toggle the color?

HowJMay, 2020-06-28:
@per1234  @rsora I will add flag `--plain` in `cli/board/details.go` and `cli/lib/check_deps.go` to print plain text instead of color text. What do you guys think?

matthijskooijman, 2020-06-29:
Maybe using `--color=[auto|always|never]`, like many GNU tools support, could be nice for consistency?

E.g. [man ls](https://www.man7.org/linux/man-pages/man1/ls.1.html) has:

>        --color[=WHEN]
>              colorize the output; WHEN can be 'always' (default if
>              omitted), 'auto', or 'never'; more info below
>
> ...
>
>        Using color to distinguish file types is disabled both by default and
>        with --color=never.  With --color=auto, ls emits color codes only
>        when standard output is connected to a terminal.

ubidefeo, 2020-06-29:
I'm with @matthijskooijman on this

if we support colour format we should use more legacy-like methods

umbynos, 2022-10-13:
I think we can close this one since we added the flag `--no-color` in https://github.com/arduino/arduino-cli/pull/1418

closed 2022-10-13 by umbynos