arduino/arduino-cli #660: "board list" command doesn't follow proxy configuration

cmaglie, 2020-04-16:
The command `arduino-cli board list` should use proxy to call board VIP/PID identification API, but this is not implemented yet: https://github.com/arduino/arduino-cli/blob/6903076cd3e0ba05791dbd248907a7d42c7f9e16/commands/board/list.go#L57

We may consider to use `downloader` for this piece too.

cmaglie, 2020-05-27:
fixed by #672

closed 2020-05-27 by cmaglie

j-d-salinger, 2023-06-24:
It is 2023, and it doesn't follow the http_proxy and https_proxy global linux variables

cmaglie, 2023-06-26:
The proxy setting must be written in the configuration file using the `network.proxy` key.

If you want to request support for `http_proxy` and `https_proxy` env vars, please open a new issue and provide a link to the env var specification.