arduino/arduino-cli #662: Board not supported message, even though it was detected

max-wittig, 2020-04-16:
## Bug Report

Board not supported message, even though it was detected

I may be doing something wrong, but I'm not sure what. Already tried a restart

### Current behavior

```sh
Port         Type              Board Name  FQBN            Core       
/dev/ttyACM0 Serial Port (USB) Arduino Uno arduino:avr:uno arduino:avr
/dev/ttyAMA0 Serial Port       Unknown                                

max@ubuntu-19-10:~$ arduino-cli board attach /dev/ttyACM0 Test.ino 
Attach board error: no supported board found at serial:///dev/ttyACM0
```

### Expected behavior
Expected sketch to be uploaded

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.9.0 Commit: e4d02b4
- OS and platform:

```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan

Linux ubuntu-19-10 5.3.0-1022-raspi2 #24-Ubuntu SMP Fri Mar 27 21:38:07 UTC 2020 armv7l armv7l armv7l GNU/Linux
```

max-wittig, 2020-04-16:
Solved it. Libraries need to be installed first.

For others:

```sh
arduino-cli core install arduino:avr
arduino-cli board attach /dev/ttyACM0 /home/max/Test
```

ubidefeo, 2020-04-18:
@cmaglie @rsora 
We really need to add these info at the very beginning of the README file.
A summed up quick-start guide.
We know very well that nobody goes to read the docs :D

max-wittig, 2020-04-18:
I guess the error message should just give a hint, if possible.

max-wittig, 2020-04-18:
But thanks for maintaining this great library. I can flash my Arduino automatically now :D