arduino/arduino-cli #664: please support file: protocol for index files and platform/tools components

PeterSommerlad, 2020-04-19:
## Bug Report

### Current behavior

`$ arduino-cli core list`

```
Error updating index: downloading index file://localhost//Users/sop/Arduino/Arduino-Cpp17/package_Cpp17AVR_index.json: Get "file://localhost//Users/sop/Arduino/Arduino-Cpp17/package_Cpp17AVR_index.json": unsupported protocol scheme "file"
```

### Expected behavior

Like the Arduino IDE arduino-cli should support the `file:` for developing cores and platforms locally. Always having to upload/download or run a local web server is time consuming. 

### Environment

- CLI version (output of `arduino-cli version`):
arduino-cli Version: 0.10.0 Commit: ec5c3ed105b32c5654fd60131a667f8557b196d5

- OS and platform:
Darwin Kernel Version 18.7.0: Mon Feb 10 21:08:45 PST 2020; root:xnu-4903.278.28~1/RELEASE_X86_64 x86_64 i386
(Macos Mojave)

### Additional context

I am working on a new AVR and ARM core with more modern C++17 internals. So far on ARM things look promising wrt code size and improved type safety.

silvanocerza, 2021-05-24:
This is solved by #1098, closing.

closed 2021-05-24 by silvanocerza

cjhdev, 2022-11-15:
This issue was not completely resolved in #1098

The index file contains package definitions, and each definition has a URL field.
The fix was applied to the index file URL, but not to the package URLs within the index file.

I have confirmed that the legacy IDE supports file URLs in the package definitions.

Should this issue be reopened?