arduino/arduino-cli #684: DEFAULT_WARNINGS_LEVEL should be "default"

nerdralph, 2020-05-05:
## Bug Report
DEFAULT_WARNINGS_LEVEL is "none"
### Current behavior
DEFAULT_WARNINGS_LEVEL is set to "none"
https://github.com/arduino/arduino-cli/blob/0866d99f71ec7c679681e14c16459d7ce9222c67/legacy/builder/builder.go#L39

### Expected behavior
const DEFAULT_WARNINGS_LEVEL = "default"

### Environment
Version: 0.10.0 Commit: ec5c3ed

### Additional context
The Arduino IDE has four preferences options for warnings: None, Default, More, All
based on the selection, "none", "default", "more", or "all" gets concatenated with compiler.warning_flags. and the corresponding option from platform.txt is used in the build flags.
https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L16

matthijskooijman, 2020-05-05:
One complication of enabling warnings by default is that you get warnings for *all* compiled code, including code you are not actually working on (e.g. warnings in the Arduino core and libraries) and not in a good position to fix. This can be especially confusing to novice users.

IMHO, having warnings enabled for the sketch, but not other parts, would be ideal.

See also https://github.com/arduino/Arduino/issues/4184 for previous discussion of this issue for the Java IDE, with some alternative proposals too.

cmaglie, 2020-05-18:
related to https://github.com/arduino/arduino-builder/pull/301

per1234, 2020-06-12:
One thing to consider is that the target user of Arduino CLI is different from the target user of Arduino IDE. Although I'm a big fan of always enabling and paying attention to compiler warnings, I can somewhat understand the reasoning behind disabling them by default in the Arduino IDE. After all, we regularly get posts on the forum:
>Something is wrong with my Arduino. It says
>```
>Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
>Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. >Maximum is 2048 bytes.
>```

and much more often than that get posts from people freaked out by innocuous warnings from libraries, thinking they are "errors".

However, that level of beginner should not be using Arduino CLI, so we don't need to cater to them.

matthijskooijman, 2020-06-12:
> One thing to consider is that the target user of Arduino CLI is different from the target user of Arduino IDE.

Fair point. However, would this work technically? Does arduino-cli have a completely separate configuration from the IDE? Once the arduino-cli is used as a backend for the IDE, would it pass all options explicitly? Would people expect to configure the cli and the IDE completely separately (especially when the IDE includes the cli)? Just some thoughts :-)

per1234, 2022-11-03:
>Does arduino-cli have a completely separate configuration from the IDE?

Yes. Arduino IDE always specifies the warning level in the compile request. That warning level is set according to the user's preferences. The default value of the preference is set in the Arduino IDE codebase:

https://github.com/arduino/arduino-ide/blob/260227e79a05fefeb87ba1400c14215b358cc20c/arduino-ide-extension/src/browser/arduino-preferences.ts#L93