arduino/arduino-cli #699: How to add required library with the code in src or some relvant folder to enable compilation

rnbokade, 2020-05-09:
Am trying to setup CI for my Arduino project on Github. My sketch uses many self-made header files and which I want to be included in /src subfolder. Tried to write #include "/src/abc.h" . However then different libraries again depend on other libraries. I don't want to change all of those libraries.  Does Arduino-cli has any option or any way in which I could compile the files as they are without changing the path like I run in desktop

per1234, 2020-05-09:
>Am trying to setup CI for my Arduino project on Github.

If it's a public repository, please post a link to it so we can get a better idea of what you're trying to accomplish.

>different libraries again depend on other libraries. I don't want to change all of those libraries.

Why would you need to change the libraries? Is it because you're bundling them with the sketch under the `src` subfolder of the sketch?

>like I run in desktop

It's not clear to me what you mean by this. Are you saying that something that works when you use the Arduino IDE doesn't work when you use Arduino CLI?

rnbokade, 2020-05-09:
I have written fec.c and fec.h and few other such header and c files which I include in my sketch with #include<fec.h> #include<abc.h> on my computer... I am able to compile my sketch because I have kept these files in /Documents/Arduino/Libraries/custom/ on my windows PC. I want to take my same code that runs on the desktop and upload it to GitHub while having CI for it. Read somewhere that when compiling sctech if you keep headers and c files in /src subfolder in your folder it compiles similarly. But it is not doing so In this case

rnbokade, 2020-05-09:
https://github.com/ArminJo/arduino-test-compile
Was following this tutorial which seemingly uses Arduino cli on docker image to do so...

per1234, 2020-05-25:
OK, I understand now. It is possible to do this using Arduino CLI. Arduino CLI allows you to place a configuration file in the sketch folder. So you can have sketch-specific configurations. This means you can define any arbitrary location you like for your user directory (this is just another term for what we call the "sketchbook directory" when using the Arduino IDE) to be installed, including in a subdirectory of the sketch! The libraries are installed in the `libraries` subdirectory of the user directory.

Let's say you have this sketch folder structure:
```
Foo
|_ Foo.ino
|_ arduino-cli.yaml
|_ user
   |_ libraries
      |_ Bar
         |_Bar.h
```
and in arduino-cli.yaml:
```yaml
directories:
  user: user
```
(You can name the user directory anything you like, I just happened to name it `user` in my example)

Then if you have in Foo.ino:
```c++
#include <Bar.h>
```
It will work just as it does if you had the Bar library installed in your global sketchbook when using the Arduino IDE.

---
There is a request for adding a formal system for doing this here: https://github.com/arduino/Arduino/issues/7755 (as well as a couple related ones I linked to from that issue).

Since we already have that feature request, and I think I've answered your question, I'll go ahead and close this. If you find that the current capabilities of Arduino CLI or the feature requests won't provide for your use case, you are welcome to reply here with more information and I'll consider re-opening the issue.

closed 2020-05-25 by per1234

matthijskooijman, 2020-05-25:
> Arduino CLI allows you to place a configuration file in the sketch folder.

Are you sure? From reviewing the code, I think it allows putting a config file in the *current working directory* (or any parent directory), which might not necessarily be the same as the sketch directory.

https://github.com/arduino/arduino-cli/blob/8220b32d158723143ff942eb57d21534dfe91d78/cli/cli.go#L154-L156

Unless the current dir is changed to the sketch directory, but I think this code runs way before figuring out what sketch to use, and I could not find any `Chdir` in the code anywhere.

per1234, 2020-05-25:
Yes, you're right. I should have been more specific about that. Thanks for clarifying.

I meant that you should place the configuration file that is used to provide the configuration specific to a sketch in that sketch's directory. This way, you are bundling the Arduino CLI configuration that is required for the sketch to work with that sketch.

I didn't mean to imply that Arduino CLI treats the configuration file in that location any differently than it would a configuration file in any other random location (though that would be a nice feature now that I think of it).

It's convenient to run arduino-cli commands from the sketch directory because this way you don't need to add the sketch path argument to your commands.