arduino/arduino-cli #701: The Arduino-builder compiler doesn't work 1.8.12 linux 64

HennieP12, 2020-05-11:
Arduino-builder doesn't work linux 64 bit, in 1.8.12.<br>Ran arduino-builder from terminal, segmentation fault.<br>Arduino itself (1.8.12) returned with error code 139. Thanks!<br>
OS:Linux Mint Cinnamon, 19.3 Tricia 64-bit
Processor: Intel Core 2 Duo T6600 @ 2.2 ghz
.desktop: exec: sudo "/home/henrik/arduino-1.8.12/arduino". (I don't feel like running commands before running arduino.)
trying to upload grbl 0.9J.
<br>Arduino app output:<br>
```
Arduino: 1.8.12 (Linux), Board: "Arduino Uno"

/home/henrik/arduino-1.8.12/arduino-builder -dump-prefs -logger=machine -hardware /home/henrik/arduino-1.8.12/hardware -tools /home/henrik/arduino-1.8.12/tools-builder -tools /home/henrik/arduino-1.8.12/hardware/tools/avr -libraries /root/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10812 -build-path /tmp/arduino_build_370773 -warnings=none -build-cache /tmp/arduino_cache_909974 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/henrik/arduino-1.8.12/hardware/tools/avr -verbose /home/henrik/grblUpload/grblUpload.ino
/home/henrik/arduino-1.8.12/arduino-builder returned 139

Error compiling for board Arduino Uno.
```
<br>
1.8.10 worked fine for what I wanted to do, 1.8.12 didn't.<br>
Can you people at Arduino fix it?

HennieP12, 2020-05-12:
Thanks, @facchinm!

cmaglie, 2020-05-18:
at the moment we produce the arduino-builder executable with:

```
GOOS=linux GOARCH=arm64 go build -o arduino-builder
```

maybe we should add the `GO386=387` flag to support a wider audience? Let me try a build with this flag...

cmaglie, 2020-05-18:
aehm... I mean:

```
go build -o arduino-builder $MAIN_GO
cp arduino-builder distrib/linux64

GOOS=linux GOARCH=386 go build -o arduino-builder $MAIN_GO
cp arduino-builder distrib/linux32
```

so neither x64 nor x86 have any CPU-specific flags.

cmaglie, 2020-05-18:
@HennieP12 
I've rebuild `arduino-builder` with the latest go1.14.3 and different set of flags, may you try them and report if anyone of them works for your system?

1. `GOARCH=amd64`:  [arduino-builder-x86_64.zip](https://github.com/arduino/arduino-cli/files/4644127/arduino-builder-x86_64.zip)
2. `GOARCH=amd64 GO386=387`: produce the same output as 1.
3. `GOARCH=386`: [arduino-builder-x86_32.zip](https://github.com/arduino/arduino-cli/files/4644129/arduino-builder-x86_32.zip)
4. `GOARCH=386 GO386=387`: [arduino-builder-x86_32-387.zip](https://github.com/arduino/arduino-cli/files/4644133/arduino-builder-x86_32-387.zip)

note that `GO386=387` seems to not affect the 64 bit version in any way that's why `GOARCH=amd64 GO386=387` is missing.

HennieP12, 2020-05-21:
Thanks! for some reason, though, the first one is a drop in replacement! (I tried to compile first with the new arduino-builder extracted from archive one, and replacing the one I got with the arduino ide, and it worked to compile!)

umbynos, 2022-10-18:
Hi @HennieP12 we are already building with `GOARCH=amd64` so your problem should already been solved. Feel free to reopen the issue if needed

closed 2022-10-18 by umbynos