arduino/arduino-cli #705: arduino-cli commands not recognized across folders

MicheleCioccarelli, 2020-05-13:
I recently downloaded the arduino-cli and i was following the official tutorial to get started, but the command line kept saying that it didn't recognize "arduino-cli"
Today i discovered that if I place the file "arduino-cli.exe" in the folder I am working on, everything is fine.
Is this intended?

ubidefeo, 2020-05-13:
@MicheleCioccarelli 
The `arduino-cli.exe` command should be reachable within your system's search path.
Because on Windows you don't have an installer but just download the `.exe` file you'll have to manually copy it it some path your system can reach easily.
You can take a look at this article (first one which came out of a google search) and it should apply.
Let me know if you sort it out.

@rsora I was surprised this bit about installing on Windows is not part of the doc :)

rsora, 2020-05-13:
I summon @per1234 to the rescue :smile_cat: 
We could also think about generating a `.msi` installer as I saw [here](https://github.com/cli/cli/blob/master/.github/workflows/releases.yml#L59)

ubidefeo, 2020-05-13:
documentation is key, but I'll leave the decision to create an `.msi` installer to you.
I don't like that much when it comes to command line tools, but it's just my opinion ;)

umbynos, 2022-10-18:
Hi @MicheleCioccarelli we now have a MSI installer :partying_face: This installer will also place the binary in the windows PATH env variable.

closed 2022-10-18 by umbynos

umbynos, 2022-10-18:
https://github.com/arduino/arduino-cli/releases/download/0.28.0-rc.1/arduino-cli_0.28.0-rc.1_Windows_64bit.msi