arduino/arduino-cli #708: panic: runtime error: invalid memory address or nil pointer dereference (version 0.10.0)

nuno-silva, 2020-05-17:
## Bug Report

### Current behaviour

```
$ arduino-cli lib list
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xb533d7]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/lib.listLibraries(0x0, 0xc000160000, 0x0, 0x145b620, 0x0)
	/__w/arduino-cli/arduino-cli/commands/lib/list.go:58 +0x37
github.com/arduino/arduino-cli/commands/lib.LibraryList(0xe99340, 0xc0000aa010, 0xc000161d40, 0x1, 0x1, 0xc00010f200)
	/__w/arduino-cli/arduino-cli/commands/lib/list.go:38 +0xb5
github.com/arduino/arduino-cli/cli/lib.runListCommand(0xc0001c4280, 0x1459b08, 0x0, 0x0)
	/__w/arduino-cli/arduino-cli/cli/lib/list.go:55 +0x110
github.com/spf13/cobra.(*Command).execute(0xc0001c4280, 0x1459b08, 0x0, 0x0, 0xc0001c4280, 0x1459b08)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0x142f620, 0x0, 0xc000c0, 0xc000094058)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fb
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:26 +0x2d
```

### Expected behavior

Expected a list of installed libs and no segmentation fault.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.10.0 Commit: ec5c3ed` (latest release)
- OS and platform: `Ubuntu 18.04.4 LTS`, `Linux 4.15.0-99-generic`

### Additional context

Installed using the `install.sh` script.

nuno-silva, 2020-05-17:
Upon further investigation, this is caused by `/home/nuno/.arduino15` being owned by `root`:

```
$ ls -la /home/nuno/.arduino15
total 12
drwxr-xr-x  2 root root 4096 mai  2 05:41 .
drwxr-xr-x 47 nuno nuno 4096 mai 17 02:48 ..
-rw-r--r--  1 root root  104 mai  2 05:41 inventory.yam
```

`sudo chown -R nuno:nuno /home/nuno/.arduino15` fixes it. However, the issue is still relevant: it should display a pretty "permission denied error". I've updated it accordingly.

bokunodev, 2021-09-23:
same problem, different solution.
its because my linux user is not part of `uucp` group.

silvanocerza, 2021-12-07:
@nuno-silva does this happen with the latest version? I can't reproduce it by setting my `.arduino15` owner to `root`.

nuno-silva, 2021-12-09:
Hi @silvanocerza ! At least the case in my original report seems to be resolved as of Version: 0.20.1 Commit: abb21449:

```sh
test@739413ccbf4a:~$ ls -l .arduino15/
total 4
-rw-r--r-- 1 root root 104 Dec  9 01:08 inventory.yaml
test@739413ccbf4a:~$ arduino-cli version   
arduino-cli-0.10.0 Version: 0.10.0 Commit: ec5c3ed
test@739413ccbf4a:~$ arduino-cli lib list
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xb533d7]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/lib.listLibraries(0x0, 0xc000160000, 0x0, 0x145b620, 0x0)
	/__w/arduino-cli/arduino-cli/commands/lib/list.go:58 +0x37
github.com/arduino/arduino-cli/commands/lib.LibraryList(0xe99340, 0xc000036068, 0xc000151d40, 0x1, 0x1, 0xc000101200)
	/__w/arduino-cli/arduino-cli/commands/lib/list.go:38 +0xb5
github.com/arduino/arduino-cli/cli/lib.runListCommand(0xc0001a6780, 0x1459b08, 0x0, 0x0)
	/__w/arduino-cli/arduino-cli/cli/lib/list.go:55 +0x110
github.com/spf13/cobra.(*Command).execute(0xc0001a6780, 0x1459b08, 0x0, 0x0, 0xc0001a6780, 0x1459b08)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0x142f620, 0x0, 0xc000c0, 0xc00009c058)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fb
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:26 +0x2d
test@739413ccbf4a:~$ sudo mv /usr/local/bin/arduino-cli /usr/local/bin/arduino-cli-0.10.0
test@739413ccbf4a:~$ sudo BINDIR=/usr/local/bin sh arduino-cli-install.sh
Installing in /usr/local/bin
ARCH=64bit
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.20.1_Linux_64bit.tar.gz
arduino-cli  Version: 0.20.1 Commit: abb21449 Date: 2021-11-29T11:34:58Z installed successfully in /usr/local/bin
test@739413ccbf4a:~$ arduino-cli lib list
Error creating instance: Failed to create downloads directory: mkdir /home/test/.arduino15/staging: permission denied
```

so I'm closing this :)
 There may be some other edge cases such as the one reported by @bokunodev though. Please reopen if appropriate. Thanks!

closed 2021-12-09 by nuno-silva